LK8000/LK8000 #868: Multitarget waypoint name not clipped in multimaps

pventafridda, 2016-04-01:
I remember not too much time ago Bruno you made a fix for main map target name clipped correctly.
Same thing is needed for multimaps..

brunotl, 2016-04-01:
cf . "Common\Source\Draw\LKDrawLook8000.cpp" line 236 - 238 and line 471 to 475

pventafridda, 2016-04-01:
ok thanks later i fix

From: Bruno de Lacheisserie 
Sent: Friday, April 01, 2016 4:14 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] Multitarget waypoint name not clipped in multimaps (#868)

cf . "Common\Source\Draw\LKDrawLook8000.cpp" line 236 - 238 and line 471 to 475

pventafridda, 2016-04-01:
No the problem is another. LKDrawLook is the same for all multimaps. The issue is around the clip region when we have a vario displayed initially.
I am fixing.

From: Bruno de Lacheisserie 
Sent: Friday, April 01, 2016 4:14 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] Multitarget waypoint name not clipped in multimaps (#868)

cf . "Common\Source\Draw\LKDrawLook8000.cpp" line 236 - 238 and line 471 to 475

pventafridda, 2016-04-02:
Argh, I am wrong. The ClipRect is simply cutting the text at the right point, already.
This is how it is working. But it is cutting the last letter, if existing, it is not removing it entirely. 
I thought we had fixed this in some other part of the code.

AlphaLima, 2019-11-26:
checked with V7.x on LINUX, PC and Android. 
No more problem at all.
CAN BE CLOSED!

closed 2019-11-26 by brunotl