LK8000/LK8000 #871: Buttons ScreenView Zoom in out very slow, RefreshMap not triggering

pventafridda, 2016-04-04:
On both Windows and Linux we are no more getting fast response to zoom buttons.
InputEvents line 1247 is forcing RenderMap but it is not enough.
No idea.

brunotl, 2016-04-05:
define "fast response" please.

pventafridda, 2016-04-05:
Click on the moving map for zoom. Response is fast.
Click on the zoom in – zoom out buttons, response is delayed.
On PC: it is faster but quick button presses are ignored (I have to check for debouncing maybe).
On OpenGL: very slow response.
Which is strange, since on moving map action is immediate.

From: Bruno de Lacheisserie 
Sent: Tuesday, April 05, 2016 11:10 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] Buttons ScreenView Zoom in out very slow, RefreshMap not triggering (#871)

define "fast response" please.

commit 083dc8d5 (2016-04-06): [fix #871] don't set MapDirty to false in MapWindow::RequestFastRefresh(), overwrite previous MapWindow::RefreshMap().
closed 2016-04-06 by pventafridda