LK8000/LK8000 #883: 23 Para/Hang Gliders Specials (PNA)

SergioDaSilva82, 2016-04-25:
I only can get 23 Para/Hang Gliders Specials menu on my PNA (Holux 61CS) when I set Engineering Menu: ON.

Clean install of 5.0A then updated to 5.3P version.

PNA model: Holux 61CS (in vertical position).

pventafridda, 2016-04-25:
i fix it thx

From: Sérgio Da Silva 
Sent: Monday, April 25, 2016 10:52 AM
To: LK8000/LK8000 
Subject: [LK8000/LK8000] 23 Para/Hang Gliders Specials (PNA) (#883)

I only can get 23 Para/Hang Gliders Specials menu on my PNA (Holux 61CS) when I set Engineering Menu: ON.

Clean install of 5.0A then updated to 5.3P version.

PNA model: Holux 61CS (in vertical position).

SergioDaSilva82, 2016-04-25:
+#define NUMOFCONFIGPAGES 23? 

Instead of NUMOFCONFIGPAGES 22?

commit 6addc3eb (2016-04-25): Fix #883 disappeared config page for paragliders specials
closed 2016-04-25 by pventafridda