LK8000/LK8000 #885: Reset comms

PhilColbert, 2016-04-30:
Maybe a small issue with Restart comms on the kobo ( only had chance to test on there ) both devices (A&B) used to come back working in a second or so, now it waits about a minute with no data coming in on either device and then a ( automatic ) restart comms message comes up and it works once more, probably a watch dog kicking in ?

Device A - tested with bluefly and generic

Device B - TCP server.

https://goo.gl/photos/6JzPgfshffc1TauF6

pventafridda, 2016-04-30:
Big issue probably. Unlucky version for kobo..
---
Sent from mobile

PhilColbert, 2016-04-30:
Not a massive issue, it does come back  in a bit of time :) not a major problem to be honest , just reporting it

ienders13, 2016-04-30:
yes same here, manual reset kills bluefly data stream until auto re-start fixes it. I'm guessing the manual reset is not accessing the same code as the auto reset, maybe not updated since your amendments.

pventafridda, 2016-04-30:
no it is ther rtscts management that is wrong, I have to regress that fix.

From: ienders13 
Sent: Saturday, April 30, 2016 1:39 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Comment 
Subject: Re: [LK8000/LK8000] Reset comms - Version 5.3q (#885)

yes same here, manual reset kills bluefly data stream until auto re-start fixes it. I'm guessing the manual reset is not accessing the same code as the auto reset, maybe not updated since your amendments.

pventafridda, 2016-04-30:
I have released a fix version for kobo on the forum
---
Sent from mobile

PhilColbert, 2016-04-30:
Thanks, just confirming the reset comms now works as before and bluefly sounds are back, thank you !

closed 2016-04-30 by pventafridda