LK8000/LK8000 #887: New Odometer

de07y, 2016-04-30:
Hello, 
We had a discussion on a french forum (parapentiste.info) about "Odometer". We are searching for a LK8000 box indicating odometer but without including circling?
It would give pilot the (close) real distance done since the take off. (Exactly the current Odometer box but without the circling distances).
Hope it would be possible one day to add this new item?

Kind Regards.
Denis

pventafridda, 2016-04-30:
quite easy to do, in fact.
Let me see, I think we can fix it by next beta.

From: de07y 
Sent: Saturday, April 30, 2016 9:12 PM
To: LK8000/LK8000 
Subject: [LK8000/LK8000] New Odometer (#887)

Hello, 
We had a discussion on a french forum (parapentiste.info) about "Odometer". We are searching for a LK8000 box indicating odometer but without including circling?
It would give pilot the (close) real distance done since the take off. (Exactly the current Odometer box but without the circling distances).
Hope it would be possible one day to add this new item?

Kind Regards.
Denis

pventafridda, 2016-04-30:
Easy fix, it is already committed for next version.

From: de07y 
Sent: Saturday, April 30, 2016 9:12 PM
To: LK8000/LK8000 
Subject: [LK8000/LK8000] New Odometer (#887)

Hello, 
We had a discussion on a french forum (parapentiste.info) about "Odometer". We are searching for a LK8000 box indicating odometer but without including circling?
It would give pilot the (close) real distance done since the take off. (Exactly the current Odometer box but without the circling distances).
Hope it would be possible one day to add this new item?

Kind Regards.
Denis

de07y, 2016-05-01:
Hello,
Rhooo what'a a great news for us this morning !
Many thanks.
I don't know if it was so easy to do it but I know you are doing a lot with all the fix added in each versions.

Best Regards!
Denis

closed 2016-05-01 by pventafridda