LK8000/LK8000 #891: Task setting - gates problem - No tsk start

polachp, 2016-05-17:
There is a problem with task definition on my KOBO + Pa6c GPS. When task is defined, after setting up gates, the overlay says NO TSK START.

Here you can find igc + nmea logs + tsk setting etc.
https://drive.google.com/file/d/0B8ddGxn3rsO-ZEJkRkp6d2J2T28/view?usp=sharing

polachp, 2016-05-23:
I made analysis on NMEA data and there are several bad sentences (checked for checksum). But I thought LK will discard bad sentences, when Device NMEA checksum is enabled.

The time is set properly in overlay, position too. So why the task is not started? 
When I turn off the checksum in device setting, I get the MASTER TIME RESET info and then the task overlay shows gates properly.

pventafridda, 2016-05-23:
For five years task gates have been working properly, so I have no idea what is wrong with the gps you are using.

polachp, 2016-05-24:
I tested it again and there SOME problem with timegates. The GPS i am using is Globaltop TIITAN 3 GLONASS module, which actually works great with XCSOAR and TOP HAT installed on same KOBO)

I have done some more testing with CONDOR connected wia wifi TCP server with NMEA log. The time gates were working when I loded the task. 
But I step into foze/crash bug several times while trying to configure timegates during the flight. The system must be restarted to work again - it was dead frozen.

I was able to reproduce this behavior several times:
1/ added 4 points from map
2/ step into NAV/taskEDIT ui
3/ setup radiuses
4/ then go to timegates => instead of timegates dialog appering, the system is dead.

I was able to reproduce SAME bug, but by trying to enter timegates dialog directly from NAV 3/3 menu.
I am trying to help. I uploaded whole lk8000 directory with NMEALOG etc.
I consider freezing biger problem than not working gates.

I will provide you more logs if you are interested in this problem. From my GPS (while the data looks completely OK, I did some NMEA validation functions in C# and there were few sentences with wrong checksum. The time in LK system is set ok after getting FIX. There is only one difference the GPS suppots GLONASS so it sends some more sentences, that LK most probably cant parser( I havent found code that do so in parse.cs)). But I dont think this is the reason for NO TSK START.

The files are to download here: https://drive.google.com/file/d/0B8ddGxn3rsO-d0lrVWx5SHBXaTA/view?usp=sharing

pventafridda, 2016-05-24:
I verified, it is a critical bug leading to crashes .
It happens when inserting a waypoint from the map with Insert in task button.
It happens on linux, it does not happen on PC.

pventafridda, 2016-05-25:
dlgTimeGates.xml prpPGOptimizeRoute has DataType boolean and should be managed by GetAsBoolean and Set .
But dlgTimeGates.cpp setVariables() is treating it as a DataType enum adding text with addEnumText.
Memory leak.
Maybe we should ensure that addEnumText is performing on a valid data structure.

The crash happened entering the dialog of TimeGates while setting the ON-OFF for OptimizeRoute.
If disabled, showing OFF , no crash. If enabled , showing ON , problem.

brunotl, 2016-05-25:
only one solution for avoid this kind of problem : never use pointer cast.

brunotl, 2016-05-25:
problem exist on all platform and all version since 12/2011

pventafridda, 2016-05-25:
Argh. There are hundreds or thousands of pointer casts now..
---
Sent from mobile

pventafridda, 2016-05-25:
Well at least I have some good time in the next evenings checking all add 
text in enumerations. I am not sure though about the start gate problem 
initially brought as an issue.
---
Sent from mobile

pventafridda, 2016-05-25:
I fix it tonight and search for other similar problems concerning dialogs.

From: Bruno de Lacheisserie 
Sent: Wednesday, May 25, 2016 11:09 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Comment 
Subject: Re: [LK8000/LK8000] Task setting - gates problem - No tsk start (#891)

problem exist on all platform and all version since 12/2011

pventafridda, 2016-05-25:
It is fixed, but tests on timegates must be performed from scratch.

brunotl, 2016-05-31:
in file you have send, i have found 2 diffrent crash : 

1 - time gate dialog, already fixed 
2 - buffer overflow due to invalid NMEA feeds in parser.

can you confirme you have disabled NMEA checksum check ?

polachp, 2016-06-06:
Tested today. Hates are working, task start working.
Entering timegates menu no longer freezes the system.

Great job!

pventafridda, 2016-06-06:
Ah good! Very good , thanks for testing!
---
Sent from mobile

closed 2016-06-06 by pventafridda