LK8000/LK8000 #913: OTG /dev/ttyACM0/ port

de07y, 2016-07-25:
Hello,
It seems the OTG /dev/ttyACM0/ port is not readable on LK8000.

A friend of mine bought a new gps device ( http://www.lebipbip.com/fr/legpsbip-solar-vocal-gps-alti-vario/ ) and he can use XCSoar but not LK8000:   OTG /dev/ttyACM0/  port cannot be found.

As I saw several messages with this problem on other forums, is it possible to add it?

My friend is using Kobo Touch device.

Hope it will help.

Kind Regards,
Denis

brunotl, 2016-07-25:
in LK8000, OTG is only supported by device we have, reason is simple, to much device bricked with XCSoar.
device currently supported are Mini, Glo, GloHD and Touch 2.0

de07y, 2016-07-25:
Ok Bruno, thanks for your answer.

brunotl, 2016-07-25:
we can add kobo Touch in supported device list , but we need device, if anyone want send one to me, i'll do it...

de07y, 2016-07-25:
Ok I put a message on the french forum . Which postal address can I use to send it if I found one?

ka-ba, 2017-06-10:
I just got one (a Kobo Touch, without "2.0") - just one, so I wont send it away. :)
But maybe we can do some testing in team-work? I activated OTG on XCSoar on it and it works just fine (can attach my XCTracer via USB), but I'd like to try LK8000 on it (don't know either program yet).

brunotl, 2017-11-03:
done

closed 2017-11-03 by brunotl