LK8000/LK8000 #961: Create new bitmaps for hires

pventafridda, 2017-01-15:
Currently all bitmaps are 20x20,  good in the range 110-180 dpi. 
New hires devices are in the range 400-500 dpi .
I would suggest going to 60x60 and use hires bitmaps still keeping lowres for old devices.
Reason: on low res we cannot split pixels and downstretching had bad results experienced in the past.
Over a certain DPI (like 250) we load the hires set of bitmaps.

ToninoTarsi, 2017-01-27:
Could a style like this be ok ?

[image]

Original one : 

[image]

pventafridda, 2017-01-27:
of course!!
Paolo

---
Paolo Ventafridda
+393339589922

ToninoTarsi, 2017-01-27:
Ok . I will try to make a set of most important for Android at 64x64 during the weekend

brunotl, 2021-03-31:
48bcbb33