LK8000/LK8000 #963: Flight color trace to be rescaled for hires

pventafridda, 2017-01-16:
Too thin on hi res,

pventafridda, 2017-01-24:
Seems good..

pventafridda, 2017-01-27:
Please check, it is important that the trace is big enough,,

brunotl, 2017-01-31:
now to big, but can be change in system config, maybe only default value are wrong
kobo  eed to be check because line width is not scaled.

[image]

pventafridda, 2017-01-31:
Maybe the latest changes on ScreenScale made them big.
When do you merge them?
How is the runway working now?

From: Bruno de Lacheisserie 
Sent: Tuesday, January 31, 2017 6:43 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Assign 
Subject: Re: [LK8000/LK8000] Flight color trace to be rescaled for hires (#963)

now to big, but can be change in system config, maybe only default value are wrong
kobo eed to be check because line width is not scaled.

brunotl, 2017-01-31:
look good on all platform if configured to 11 instead of 16 ...

pventafridda, 2017-01-31:
What change 11 to 16? Where?

ToninoTarsi, 2017-01-31:
Just a curiosity : What was 16 for ? 16 pixel ? 16 mm ?
If it was only a calibrated value now the new one could be (16/16)*11 :-)  ... maybe.

brunotl, 2017-01-31:
pixel because before commit 9b8a4f3a939303e92a86dd183407462889a87b30 snailtrail does not take account screen scaling ...

pventafridda, 2017-01-31:
I will fix it as soon as I set up my android compile by the weekend.
Meanwhile it is worthless to change dimensions and sizes until the new scaling system is settled.

brunotl, 2017-02-02:
sceenshot is windows ...

pventafridda, 2017-02-02:
I know. today I work on it. I have discovered the problem. screenscale is wrong obviously cannot be used for rescaling without loss of accuracy.
Solution is to use IBLSCALE.
Rescale pixel is also wrong. because we should rescale only in reference to a 5 inches screen,  after IBLSCALE. this is the point .

brunotl, 2017-02-02:
this why i have fixed all ScreenScale usage ...

pventafridda, 2017-02-02:
I agree screenscale must not be used like that . when I am back home I 
write a more detailed explanation of what I have thought.

pventafridda, 2017-02-06:
I take it for fixing

pventafridda, 2017-02-07:
The problem is in the IBLSCALE . I suspect old NIBLSCALE was rounding the integer to the nearest border.
So 3.7 becomes 4.  In current scaling, we round to the lowest border. 
for niblscale 1, on dpi 330..   IBLSCALE(1)=3 *16 = 48   should be 4*16=64 .
Huge difference.

Not good to fix LKObjects.

closed 2017-02-15 by pventafridda