LK8000/LK8000 #965: PC version, problem with topology font size

pventafridda, 2017-01-17:
Not working..

brunotl, 2017-01-17:
?

pventafridda, 2017-01-18:
I am checking the latest master on PC. I am not drunk, I havent smoked anything special, and Topology font is wrong.

pventafridda, 2017-01-18:
[image]

pventafridda, 2017-01-18:
It's like the font has been made for another resolution . If you rescale in page 12, it does not work too correctly. Font topology -8 should be very very small, instead it is:
[image]

and notice that text is overlapping.

pventafridda, 2017-01-18:
I think the font is not used, for some reason. It worked in 6.0b for sure. Current master on PC is not using the correct font for topology, apparently.

pventafridda, 2017-01-18:
Probably this is the only possible explanation https://github.com/LK8000/LK8000/commit/3606a44be18c8c6c99fb34e94baf3ea4ef8fcd7a

AlphaLima, 2017-01-18:
I  worked with Fonts in FAI sectors. Will Check tomorrow

brunotl, 2017-01-18:
fixed

pventafridda, 2017-01-18:
Many thanks!

closed 2017-01-18 by pventafridda