LK8000/LK8000 #971: Aircraft icons still too small

pventafridda, 2017-01-22:


ToninoTarsi, 2017-01-22:
Yes. To small in my 530dpi phone and also on kobo hd (300dpi). May I suggest to also add  a black glider with inverted colors that will be useful on kobo.. and may be also on Android

pventafridda, 2017-01-22:
yes right

From: Tonino Tarsi 
Sent: Sunday, January 22, 2017 11:33 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Assign 
Subject: Re: [LK8000/LK8000] Aircraft icons still too small (#971)

Yes. To small in my 530dpi phone and also on kobo hd (300dpi). May I suggest to also add a black glider with inverted colors that will be useful on kobo.. and may be also on Android

pventafridda, 2017-01-22:
Resizing with RescalePixel or NIBL is not a good idea when dealing with very small values with no tolerances such as -5 9 0 -5 .. a value of 13 or 14 make a big difference in aspect then.
Easy and good result using different shapes and choosing by default the best one .

pventafridda, 2017-01-23:
Better to check with latest commit if shapes are resized a bit bigger than before.
I am afraid that for some shapes such as the triangle they were originally made too small, and we should enlarge the vectors.

pventafridda, 2017-01-24:
OK best solution is to have also bigger shapes, and let users choose.

pventafridda, 2017-01-27:
On DITHER devices glider is inverted. By now it seems ok.

closed 2017-01-27 by pventafridda