LK8000/LK8000 #977: Android: Grid out of place in MM1 (Glitch)

parapenT1sta, 2017-01-27:
In Android version when switching to MM1 (or resizing the multimap) the grid is out of place but goes back to where is suppose to be.

[image]
[image]

parapenT1sta, 2017-02-06:
I tested with the latest beta version available and still the same :(

pventafridda, 2017-02-06:
not for me, it is fixed

parapenT1sta, 2017-02-06:
https://goo.gl/photos/oueuhJ6ofRkAPnTe9

Google Nexus 6P

pventafridda, 2017-02-06:
It happens when reducing the top window opening the sideview.
It is only a glitch, due to different timing in drawing and updating some variable (1s latency).
Existing since v4 or v5, not worth to spend time on this issue really with all the things to do yet.

parapenT1sta, 2017-02-27:
I also checked that it happens running LK8000 in Nimbus

pventafridda, 2017-02-27:
it happens since 2014. since ever we made multimaps and on all platforms. 
it is the most unimportant problem we have .

closed 2017-08-14 by parapenT1sta