LK8000/LK8000 #983: Create new demo maps hires for android.

pventafridda, 2017-01-31:


brunotl, 2017-02-03:
we can use this : http://www.bware.it/listing/LKMAPS/EUR/DEMO.DIR/DEMO_250.DEM

pventafridda, 2017-02-03:
I confess I have been thinking about using a territory at Saint-Auban with 90 DEM... but I miss all waypoints there and airspaces.
The idea for demo is to have a nice looking landscape, and having the lakes around so far looks nice on screen for a demo.
Better than a flatland with nothing to show, I mean.

But yes we have 250 DEM ready for demo as well.
Can we use a demo set for android and kobo, using different maps from the rest of distribution?

From: Bruno de Lacheisserie 
Sent: Friday, February 03, 2017 12:43 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Assign 
Subject: Re: [LK8000/LK8000] Create new demo maps hires for android. (#983)

we can use this : http://www.bware.it/listing/LKMAPS/EUR/DEMO.DIR/DEMO_250.DEM

ToninoTarsi, 2017-02-03:
I think current demo area  is a very nice area also for Android ... lakes, rivers . lot of airport ...

pventafridda, 2017-02-05:
The DEMO90.DEM looks good on android, but on CE it cant be used for memory size, and on all devices if the screen is not hires (I suppose >300dpi) there are unblurred artifacts at low zoom . On PC artifacts a 7.5km. 
But Ok, I suggest to use it as a demo only on android. We can name it DEMO90.DEM and change the code to use in place of DEMO.DEM.  OR, we can copy in Maps DEMO90. 

Tony?

pventafridda, 2017-02-05:
The artifacts are created by the shading effect.
I am convinced that at at low zoom, lower shading parameters should be adopted.
Right now we stop shading at 15km zoom (RealScale10.71) .
It is a bad solution, I try to work on it.

ToninoTarsi, 2017-02-05:
For DEMO I think current 250 is ok as very few people will use it for flying ( only the one in your area) and maybe is not worth to charge additional byts to all the other. 
On phone with barometric sensor DEM is important for HG/PG as HNQ is fixed on terrain and in takeoff area (mountains 250 meters DEM can have a significant error). 
So. I will fly with 90 meters that is ok and with the DEM generator everybody can have, if he want, a 90 meters DEM. 
So I do not see any issue here but....any decision is ok. 
Ps:if we use demo90 on Android also APK will be much bigger.

pventafridda, 2017-02-05:
true. it becomes much bigger. the demo is not for flying only to show the 
software running at first install. 250 is OK I agree. we go 250 for 
android. I add demo250.dem to distribution.

pventafridda, 2017-02-05:
using baro altitude is not accurate in any case because it changes during 
the day and most important it changes right when you want to land quickly: 
a thunderstorm arriving. Much better not to use it in my opinion.

closed 2017-02-15 by pventafridda