LK8000/LK8000 #987: Landable Waypoint Icon : don't work on Linux, PC and CE for screen resolution over 480*272

brunotl, 2017-02-02:


pventafridda, 2017-02-02:
I am doing all tests with RESCALE PIXEL on for all platforms. I fix it as quick as I can, work in progress with rescaling.

brunotl, 2017-02-03:
seem's fixed if we replace line 212
```
  irw_radius=RescalePixelSize(irw_radius);`
```
by
```
#ifdef RESCALE_PIXEL
  irw_radius=RescalePixelSize(irw_radius);
#else
  irw_radius=IBLSCALE(irw_radius);
#endif
```

pventafridda, 2017-02-03:
sure but I remove the RescalePixelSize is damn wrong, it must not be used this way.

From: Bruno de Lacheisserie 
Sent: Friday, February 03, 2017 12:41 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Comment 
Subject: Re: [LK8000/LK8000] Landable Waypoint Icon : don't work on Linux, PC and CE for screen resolution over 480*272 (#987)

seem's fixed if we replace line 212

  irw_radius=RescalePixelSize(irw_radius);`
by

#ifdef RESCALE_PIXEL
  irw_radius=RescalePixelSize(irw_radius);
#else
  irw_radius=IBLSCALE(irw_radius);
#endif

pventafridda, 2017-02-03:
it is fixed.

closed 2017-02-03 by pventafridda