LK8000/LK8000 #995: DrawTerrain : dtquant max value

brunotl, 2017-02-04:
max value for dtquant is 3, that can be to small for HD or QHD screen
test requiered for look if whe can use bigger value on android target.

pventafridda, 2017-02-04:
bogomips for that. very basic . I prefer max speed than low dtquant. also 
less battery. less heating..

brunotl, 2017-02-06:
exactly, already done for some kobo model, scalling "x2" or "x4" are faster than "x3" on memory canvas.

pventafridda, 2017-02-06:
Question, dtquant too small for QHD you say. Why too small? Artifacts or too much cpu spent?

brunotl, 2017-02-06:
I just say "maybe" for map drawing time, i don't have device with QHD, i can't make test.

pventafridda, 2017-02-15:
Assuming single CPU thread for drawing, our reference was around 120-150 terrain loads /Mhz.
I think we can go up to 200 terrain loads /mhz on multicores.
This bring us to FULLHD dtquant 3, QHD dtquant 4 without any problem and no need to do oversampling for blurr.   FULLHD dtq 3 is equivalent to QHD dtq 4 .
Another matter is real vision in flight, because if you look at the screen from 30-40cm, the real size of the screen is reduced, and DPI is much bigger. Over a certain "perceived dpi" it is useless to render terrain. It is like printing with font size 8 and looking at the papaer sheet from half a meter, resulting in invisible text.

brunotl, 2017-02-15:
- Number of CPU core change nothing for draw terrain.
- blur is disabled on Greyscale and openGL target (linux, kobo, android).
- thinking only in term of CPU speed is big mistake, memory latencies are also important.

pventafridda, 2017-02-15:
That's why I said single thread.
going from FHD to QHD has only screen size in difference, memory latencies and cpu are correlated to the amount of terrain loads. 
If we want to reduce dtquant: first, we can let user choose (it is their device after all..), second we can suggest a dtquant (let's call it Terrain Quality). 
We pick up a desired screen density for optimal vision at a reference distance (say 30cm). We rescale dtquant for that density, for flight usage.
Something to keep in mind is that objects that are moved away from our eyes get smaller in size with a ratio of 1/sin .

brunotl, 2017-02-15:
suject of this issue is not about lower dtquand value, it's about bigger value ...

pventafridda, 2017-02-15:
Sorry I meant to reduce terrain loads (by rising dtquant) I thought it was implicit.

brunotl, 2017-02-15:
in your previous comments i read 
> I think we can go up to 200 terrain loads /mhz on multicores.

and
> If we want to reduce dtquant ...

so, like you says, i think max value 3 is to small for high density screen, in theory we can rise it to 5 and use real screen density instead of screen height ratio for calc value.

pventafridda, 2017-02-15:
On V2,  we have a ratio of 90.6 terrain loads *Mhz. On a very fast CE machine like MiniMap 800mhz we have 120 terrain loads *Mhz, and it is instant drawing at 800x480. This is why I think we can go "up to"  200 with confidence. With 1.4Ghz cpu,  280.000 loads. For example, FullHD at 1.4Ghz dtquant 3 is 230.000, quite near, dtquant 4 means 129 loads, underloading cpu. So we go for 3.
On QHD, dq=3  409.000 way too much. dq=4 230.000 , and it seems ok. dq=5 147.000 underloading.

pventafridda, 2017-02-22:
With sysop mode we can change dtquant and check for results.
But using Mhz as a parameter for scaling is unrealistic. BogoMips also unrealistic.
We may rescale dtquant for screen density but in a reasonable range. Maybe it is worthless.

pventafridda, 2017-03-14:
I have written the functionality to control automatically dtquant.
It can adjust lowering or rising the dtquant depending on system performance.
After tests, normally the difference is hardly visible.
I don't think it is worth using it.

closed 2017-03-14 by pventafridda