LK8000/LK8000 #1002: Report from user crash on waypoint delete android

pventafridda, 2017-02-05:
I have tested. working on V6, working on current master on PC , linux and android.
it could be a peculiar problem on that device. I cannot replicate it .

brunotl, 2017-02-06:
crash callstack : 

0  libc.so                        0x7fb58a868c tgkill
1  libc.so                        0x7fb58a6014 pthread_kill
2  libc.so                        0x7fb585f08c raise
3  libc.so                        0x7fb5859998 abort
4  libc.so                        0x7fb585cde0 (Missing)
5  libc.so                        0x7fb58a77c8 __stack_chk_fail
6  libLK8000.so                   0x7f99aab72c (Missing)
7  libLK8000.so                   0x7f999b18e0 WindowControl::OnPaint(LKSurface&, PixelRect const&)
8  libLK8000.so                   0x7f999b67e4 LKWindow<ContainerWindow>::OnPaint(Canvas&)
9  libLK8000.so                   0x7f99955930 WindowList::Paint(Canvas&)
10 libLK8000.so                   0x7f99956014 ContainerWindow::OnPaint(Canvas&)
11 libLK8000.so                   0x7f999b67f0 LKWindow<ContainerWindow>::OnPaint(Canvas&)
12 libLK8000.so                   0x7f99955930 WindowList::Paint(Canvas&)
13 libLK8000.so                   0x7f99956014 ContainerWindow::OnPaint(Canvas&)
14 libLK8000.so                   0x7f999b67f0 LKWindow<ContainerWindow>::OnPaint(Canvas&)
15 libLK8000.so                   0x7f99955930 WindowList::Paint(Canvas&)
16 libLK8000.so                   0x7f99956014 ContainerWindow::OnPaint(Canvas&)
17 libLK8000.so                   0x7f999b67f0 LKWindow<ContainerWindow>::OnPaint(Canvas&)
18 libLK8000.so                   0x7f99955930 WindowList::Paint(Canvas&)
19 libLK8000.so                   0x7f99956014 ContainerWindow::OnPaint(Canvas&)
20 libLK8000.so                   0x7f9995255c LKWindow<SingleWindow>::OnPaint(Canvas&)
21 libLK8000.so                   0x7f9995664c TopWindow::Refresh()
22 libLK8000.so                   0x7f99953020 EventLoop::Get(Event&)
23 libLK8000.so                   0x7f999b2aa0 WndForm::ShowModal()
24 libLK8000.so                   0x7f99aaa420 dlgWayPointSelect(double, double, int, int)
25 libLK8000.so                   0x7f99a8f414 (Missing)
26 libLK8000.so                   0x7f999b373c WndButton::OnLButtonUp(RasterPoint const&)
27 libLK8000.so                   0x7f999b66c4 LKWindow<ContainerWindow>::OnMouseUp(short, short)
28 libLK8000.so                   0x7f999b6698 LKWindow<ContainerWindow>::OnMouseUp(short, short)
29 libLK8000.so                   0x7f999b6698 LKWindow<ContainerWindow>::OnMouseUp(short, short)
30 libLK8000.so                   0x7f999b6698 LKWindow<ContainerWindow>::OnMouseUp(short, short)
31 libLK8000.so                   0x7f99952404 LKWindow<SingleWindow>::OnMouseUp(short, short)
32 libLK8000.so                   0x7f999b2b78 WndForm::ShowModal()
33 libLK8000.so                   0x7f99a85fa0 dlgConfigurationShowModal(short)
34 libLK8000.so                   0x7f999ab410 SystemConfiguration(short)
35 libLK8000.so                   0x7f9998a30c InputEvents::processButton(unsigned int)
36 libLK8000.so                   0x7f99977928 MenuButton::OnLButtonUp(RasterPoint const&)
37 libLK8000.so                   0x7f99977624 LKWindow<Window>::OnMouseUp(short, short)
38 libLK8000.so                   0x7f99952404 LKWindow<SingleWindow>::OnMouseUp(short, short)
39 libLK8000.so                   0x7f99958170 TopWindow::RunEventLoop()

pventafridda, 2017-02-06:
buffer overflow, fixed. dlgWayPointSelect need a deep, careful, look. sTmp used everywhere without check.

closed 2017-02-06 by pventafridda