LK8000/LK8000 #1016: Support for android eink devices.

pventafridda, 2017-02-09:
Yotaphone for example. We need dithering configurable, or something like.

ToninoTarsi, 2017-02-09:
One of my friends did already test the beta with a yota and he said was working  fine....

pventafridda, 2017-02-09:
yes it works fine but for eink we made it much better.

brunotl, 2017-02-09:
without device we can't do anything...

ToninoTarsi, 2017-02-09:
I could just compile with DITHER=y and send him a binary for test.

brunotl, 2017-02-09:
loose of time, yotaphone have 2 Screen, color and EInk, runtime detection of used screen is needed.

parapenT1sta, 2017-02-09:
You can add support for Onyx T68

pventafridda, 2017-02-09:
I don't remember if it is possible to compile for grey shades and DITHER . 
it is in the readme of the master.
I have used Linux build like that for testing dither changes

AlphaLima, 2017-02-27:
In Germany you can buy Yotaphones for 125€. So I did!
http://www.ebay.de/itm/YOTAPHONE-4G-2-Screen-Smart-Phone-Schnelle-Gebuhr-drahtloses-Aufladen-32GB-Handy-/222345784947?hash=item33c4d76e73:g:SMwAAOSw44BYUVHf
really a good quality and fast phone with eInk. pricewise better than any PNA or Kobo. A cool toy!

pventafridda, 2017-02-27:
there was a thread somewhere about bugs in that device that made me drop 
the idea of buying one.. good to know thanks

parapenT1sta, 2019-01-29:
@brunotl I think this can be closed.

parapenT1sta, 2019-01-29:
https://github.com/LK8000/LK8000/commit/280c8d53515ff11e8e2a9b2325dac27b2acda8b0

ToninoTarsi, 2019-01-29:
Unfortunately not. Commit 280c8d5 was only the beginning.
We have 2 more problems with Android e-ink devices (first simple and boring, second difficult and challenging): 
1) Replace all #ifdef DITHER by IsDither() so we can detect automatically some e-ink device . I did half of this boring work
2) We have problem with the Edp Controller and if A2 mode is not selected LK8000 does not refresh. I tryed integrating some EDP SDK ( I tryed Onix Boox) but it looks likes changing EDP mode in code does not work . I suspect root is needed. Alternatively maybe we can modify our  SurfaceView  .. I don't know at the moment.

parapenT1sta, 2019-01-29:
Thanks for the clarification.

closed 2019-11-02 by brunotl