LK8000/LK8000 #1020: Initial label don't pop out

aledand, 2017-02-10:
The initial label "GPS no fix ..." does not pop out even when the god is fixed and everything else is working fine.

Android 7.0
Moto Z

brunotl, 2017-02-10:
Please post runtime.log

aledand, 2017-02-11:
The runtime log. In which effectively seems to be logged no-fix. Is the no-fix and the label linked with the number of satellites in use ? To me seems that the position is received but strangely the number of satellites says zero. A problem in reading that part of the sentence with this phone ?

ToninoTarsi, 2017-02-11:
Is your phone running fine with other software like XCSoar, XCTrack ? Looks like your phone does not receive notifications for onLocationChanged().

aledand, 2017-02-11:
Yes I checked. Even with xcs satellites are zero but status is 3DFix. I'd a look to internalgps and the only place where apparently the location is set is also setting the status to 2.  So or it is ignored or somewhere else it set back to 1 ... like on the broadcast receiver.

Btw I guess (hope) that it is not a common issue, so better spent time with next features. As soon as back ahome I'll mount src and try to have a look by my self.

Grazie ;-)

aledand, 2017-03-07:
in the end had time to look into that.
as I guessed the issue is at
DrawGPSStatus.cpp:64 where is used "|| (DrawInfo.SatellitesUsed == 0)"
and for reasons I've not yet determined on my phone the "extras.getInt("satellites", -1)" says always zero.
Commenting the SatellitesUsed evetything works like a charm. tested on the field that "GPS No Fix" is  drawn only when needed.
Google API docs says that "All locations generated by the LocationManager are guaranteed to have a valid latitude, longitude, and timestamp (both UTC time and elapsed real-time since boot), all other parameters are optional." so to remove the redundant check should be safe.

pventafridda, 2017-03-07:
Yes I perfectly knew that we look for satellites number and not only for valid fix.
That was made for PNA problems with faulty gps.

Now that you mention android specs, we can change it for android.
Would you like to take the challenge and commit the change 
#ifdef ANDROID
...
#else
..
#endif

or we can do it for you. 

thx
Paolo

From: aledand 
Sent: Tuesday, March 07, 2017 11:43 PM
To: LK8000/LK8000 
Cc: Subscribed 
Subject: Re: [LK8000/LK8000] GPS No Fix never disappear (#1020)

in the end had time to look into that.
as I guessed the issue is at
DrawGPSStatus.cpp:64 where is used "|| (DrawInfo.SatellitesUsed == 0)"
and for reasons I've not yet determined on my phone the "extras.getInt("satellites", -1)" says always zero.
Commenting the SatellitesUsed evetything works like a charm. tested on the field that "GPS No Fix" is drawn only when needed.
Google API docs says that "All locations generated by the LocationManager are guaranteed to have a valid latitude, longitude, and timestamp (both UTC time and elapsed real-time since boot), all other parameters are optional." so to remove the redundant check should be safe.

aledand, 2017-03-07:
ok I'll do that ... better you focus on features ;-)

brunotl, 2017-03-07:
few other device don't send Satellites count (XCTracer, WinCE internal GPS, ..)

i have solved this problem by setting SattelitesUsed to -1, like that we manage the problem only where is happen.

so right fix is inside InternalSensors.cpp line 253 if n_sattelites == 0 set GPS_INFO.SatellitesUsed to -1

brunotl, 2017-03-07:
i forget : thanks for your help ;-)

pventafridda, 2017-03-07:
Yes, thanks!
Maybe you want to help on other issues...  

From: Bruno de Lacheisserie 
Sent: Wednesday, March 08, 2017 12:40 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Comment 
Subject: Re: [LK8000/LK8000] GPS No Fix never disappear (#1020)

i forget : thanks for your help ;-)

aledand, 2017-03-08:
not an expert of the LK8000 code, so sorry if going to say something silly.

in DrawGPSStatus.cpp::20 we have "DrawInfo.SatellitesUsed != 0"; with the -1 forcing should also this be changed in "DrawInfo.SatellitesUsed > 0" ?

cheers

pventafridda, 2017-03-08:
ah, no I would say it is correct.
if we have a gps and there is a valid fix and number of sats is not zero then if needed print the lock mode status on screen, otherwise simply return (do not print no valid fix..)

pventafridda, 2017-03-08:
Just for not misunderstanding, in the end who is doing the fix ?

aledand, 2017-03-08:
i can commit it

pventafridda, 2017-03-14:
Can I close this?

aledand, 2017-03-14:
Hi, by what I see change proposed by Bruno has not been applied. I tried to commit via GitHub for desktop but probably did not worked or is pending somewhere because I'm not a memeber. I guess I'd fork and pull a request ... for 1 line would faster if done by the team :-)

brunotl, 2017-03-14:
i will do it tonight

pventafridda, 2017-03-14:
sure no problems . i will commit the change for you

closed 2017-03-31 by brunotl