LK8000/LK8000 #1023: Glitch: dialogs geometry not considering caption

pventafridda, 2017-02-14:
Example, dlgConfigDevice, using 320x240. CLOSE button is at 196, h=28, there should be space enough before bottom of the screen. Instead it is painted at the very limit, exceeding screen sometimes.
Reason: the caption line (the title of the window in dialogs) is not accounted.
We need a global variable like CaptionVSize to be used handy.

AlphaLima, 2017-02-14:
Why did you remove the status line, it is a very good help for debugging 
multiple NMEA sources.

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

AlphaLima, 2017-02-14:
Oh, it is not removed, it is simply not re-enabled after comming from a 
disabled device.

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

pventafridda, 2017-02-14:
not intentional sorry i fix it

AlphaLima, 2017-02-14:
No, my fault already commited a glitch fix!

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

pventafridda, 2017-02-14:
all right! thx

brunotl, 2018-09-11:
@AlphaLima can you check if it's fixed already in 6.1t ?

AlphaLima, 2018-09-11:
Already checked it again,
yes it is fixed!

[image]

closed 2018-09-11 by brunotl