LK8000/LK8000 #1025: Screen orientation on Android

ToninoTarsi, 2017-02-15:
**New feature for Android** : At the moment screen orientation for Android is only based on internal sensor.
Especially for PG, with the instrument horizontally on the cockpit, sometime screen turns during take-off  ( it happened to me las t  Saturday during take-off ). Not all phone ( like mine ) can fix rotation in landscape mode. .. so ... something is missing.
Code ( both cpp and java ) is ready for implement  this but we do not have a setting for it ... 
Where to put it ? System config page 10 ? Screen modes ? ...

pventafridda, 2017-02-15:
system configuration I would suggest.

pventafridda, 2017-02-15:
maybe it is better to make it configurable like this:
screen rotation
always enabled
enabled only in SIM mode
always disabled

pventafridda, 2017-02-15:
but I wonder how we can rotate to landscape if it is disabled.

ToninoTarsi, 2017-02-15:
System lock is disabled..  not app lock.
For option in current code we have   :
Automatic (current ad default ) 
Landscape
Portrait
Inverted landscape 
Inverted potrait (not working in all systems)

pventafridda, 2017-02-15:
ok great

From: Tonino Tarsi 
Sent: Wednesday, February 15, 2017 4:49 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Comment 
Subject: Re: [LK8000/LK8000] Screen orientation on Android (#1025)

System lock is disabled.. not app lock.
For option in current code we have :
Automatic (current ad default )
Landscape
Portrait
Inverted landscape
Inverted potrait (not working in all systems)

ToninoTarsi, 2017-02-15:
Ok. Next week I can work on it adding the option to system menu 10

Sent from mobile

ToninoTarsi, 2017-02-19:
Closed with merge #1034

closed 2017-02-19 by ToninoTarsi