LK8000/LK8000 #1046: Airspace dialogs on OpenGL

pventafridda, 2017-02-25:
http://www.postfrontal.com/forum/topic.asp?TOPIC_ID=8498

brunotl, 2017-02-26:
I can't replicate, more info requiered ...

pventafridda, 2017-02-26:
Argh.. for me it is well replicated at first try.
load demo airspace
display airspaces on screen . maybe optional
long click on map near to airspace, outside of it
on demo I get selection of near places: CTR C BERgAMO and Valbrembo airport
i click on CTR airspace, it becomes yellow
Then I click on SELECT.

What happens: there screen keep changing between main map and the dialog  page in a 3 seconds cycle.
Second A: Map, Second B: dialog with yellow highlighted CTR C BERGAMO, Second Cialog without highlighted CTR C:
Back to map.
When on dialog page, buttons not working.
To exit the loop, change multimap when on the map page.

From: Bruno de Lacheisserie 
Sent: Sunday, February 26, 2017 1:10 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Author 
Subject: Re: [LK8000/LK8000] Airspace dialogs on OpenGL (#1046)

I can't replicate, more info requiered ...

parapenT1sta, 2017-02-26:
I tried and I couldn't replicated it. I am running Android 7.1.1 on Nexus 6P.

pventafridda, 2017-02-26:
I am running 6.0.1 on S5 NEO.
If you cant replicate  I will later tonight check on configuration.

From: Sérgio Da Silva 
Sent: Sunday, February 26, 2017 2:19 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Author 
Subject: Re: [LK8000/LK8000] Airspace dialogs on OpenGL (#1046)

I tried and I couldn't replicated it. I am running Android 7.1.1 on Nexus 6P.

parapenT1sta, 2017-02-26:
I did profile reset to make sure and I still can't replicated it in Fly and
Sim mode but I will keep trying it.

pventafridda, 2017-02-26:
it happens on android 6 Huawey P9 Lite

From: Sérgio Da Silva 
Sent: Sunday, February 26, 2017 3:59 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Author 
Subject: Re: [LK8000/LK8000] Airspace dialogs on OpenGL (#1046)

I did profile reset to make sure and I still can't replicated it in Fly and
Sim mode but I will keep trying it.

pventafridda, 2017-02-27:
It is a bad general bug involving SetTimerNotify in all dialogs. Existing also in other dialogs. I have fixed .

brunotl, 2017-02-27:
are you sure it's related to timer ?

pventafridda, 2017-02-27:
if you compile  a beta I can tell you quickly

pventafridda, 2017-02-28:
Now we know it is not related to timer. It is related to the airspace dialogs that are popped up.

pventafridda, 2017-02-28:
Some more details.
The problem happens (also) inside dlgMultiSelectList.cpp  after calling dlgMultiSelectListDetailsDialog.
Same problem if called from OnMultiSelectLisListEnter or OnEnterClicked.
The 3 seconds cycle (map-dialog-dialog with highleghted Select button) is quite strange.
In fact after showing again the old dialog, the following second is showing the SELECT button highlighted like if someone had clicked on it.

I could notice a similar behaviour in Sysop Mode after resizing the screen. Old dialog for text entry still active in the rest of the page, and it is ok. But it actually keeps looping in a 3 seconds cycle, showing the dialog without keypresses, and then with the semi-last key pressed and then the very last key pressed.
Example: from Pilot name I enter RES0800x0400 and then OK. 
The dialog will show text keyboard with no highlighted keys, then with 0 highlighted, then with OK button highlighted.

brunotl, 2017-02-28:
put break point in dlgAirspaceDetails() and look call stack ...

pventafridda, 2017-03-07:
Latest master does not have the problem anymore. If the problem is inside libraries, we need to create another issue. This particular problem is now solved.

pventafridda, 2017-03-14:
Also in 6.1e problem is gone. Closing.

closed 2017-03-14 by pventafridda