LK8000/LK8000 #1055: track value seems never over 180 when logger replay @android version

parapenT1sta, 2017-03-02:
I got this report in PGForum:

_There is a question that track value seems never over 180 when logger replay @android version. 
I'm not sure whether it is correct in fly mode. 
Track value is correct with lk8000 6.0b @PC & WinCE which I confirmed.

http://www.paraglidingforum.com/viewtopic.php?p=p522128#p522128

[image]

pventafridda, 2017-03-03:
Only on Android? Hardly possible.
Can you please check yourself ? If you can replicate I will check the code.

brunotl, 2017-03-03:
in screenshot, value is abs value of Track bearing in  [-180 : 180] range.

2 bug behind that
first is already fixed by commit a22bdf0892ea88dc8e531b6073e32734ddb90e4c
second is old bug line 569 of LKProcess.cpp, but is hidden by first fix in 6.1d

same as issue #1017

closed 2017-03-06 by brunotl