LK8000/LK8000 #1058: Crash on declaration to LX Nano via BT on Android

pventafridda, 2017-03-06:
Replicated all the times.

[686312772] ------------------------------------------------------------
[686312772] . Starting LK8000 v6.1d Feb 28 2017 LINUX
[686312772] . System Name:    Linux localhost
[686312772] . Kernel Version: 3.10.86-g76ded66
[686312772] . Kernel Build:   #2 SMP PREEMPT Mon Aug 8 11:36:11 CST 2016
[686312772] . Machine Arch:   aarch64
[686312772] . Built with GCC 4.2.1 
[686312772] . Compiler options:
[686312772]     + ENABLE_OPENGL
[686312772]     + HAVE_GLES
[686312772]     + NO_DASH_LINES
[686312772] . TESTBENCH option enabled
[686312772] . BUGSTOP option enabled
[686312772] . USELKASSERT option enabled
[686312773] . Globals_Init
[686312774] . Free ram=206484 K  storage=4294967295 K
[686312774] . Host and Cpu informations not available
[686312776] . Program execution path is <>
[686312776] . Program data directory is </storage/emulated/0/LK8000/>
[686313298] ... ProfileResetDefault
[686313298] ... LoadProfile </storage/emulated/0/LK8000/_Configuration/DEFAULT_AIRCRAFT.acf>
[686313298] ... LKProfileInitRuntime
[686313298] . No special regsets for this device
[686313298] ... LoadProfile </storage/emulated/0/LK8000/_Configuration/DEFAULT_PILOT.plt>
[686313299] ... LKProfileInitRuntime
[686313299] . No special regsets for this device
[686313299] ... LoadProfile </storage/emulated/0/LK8000/_Configuration/DEFAULT_DEVICE.dvc>
[686313299] ... LKProfileInitRuntime
[686313299] . No special regsets for this device
[686313299] ... LoadProfile </storage/emulated/0/LK8000/_Configuration/DEFAULT_PROFILE.prf>
[686313301] ... LKProfileInitRuntime
[686313301] . No special regsets for this device
[686313301] ..... WindowResize RECT (rltb) 1920 0 0 1080
[686313301] . Create main window
[686313303] ... First ScreenHasChanged: 1920 x 1080
[686313303] . InitLKScreen: AUTORES 1920x1080
[686313304] ..... ScreenSizeX      = 1920
[686313304] ..... ScreenSizeY      = 1080
[686313304] ..... ScreenDensity    = 403
[686313304] ..... ScreenGeometry   = 4
[686313304] ..... ScreenSize(enum) = 0
[686313304] ..... Screen0Ratio     = 3.970588
[686313304] ..... ScreenScale      = 4.500000
[686313304] ..... ReferenceDpi     = 110
[686313304] ..... ScreenPixelRatio = 3.663636
[686313304] ..... UseHiresBitmap   = true
[686313304] ..... ThinSize         = 4
[686313304] ..... NIBLSCALE(1)     = 4
[686313304] ..... NIBLSCALE(1.0)   = 4.500000
[686313304] ..... NIBLSCALE(2)     = 9
[686313304] ..... NIBLSCALE(2.0)   = 9.000000
[686313304] ..... GestureSize      = 225
[686313304] ..... LKVarioSize      = 120
[686313304] ..... AircraftMenuSize = 140
[686313304] ..... CompassMenuSize  = 216
[686313304] ..... TerrainQuantiz.  = 3
[686313304] . Init_Fonts_1
[686313406] . Init_Fonts_2
[686313406] ... (LKFonts) Forcing font resize
[686313700] ... Loading Fixed Bitmaps
[686313717] ... Loading Profile Bitmaps
[686313719] ... LKObjects_Create
[686313719] . SNAIL[0]=29 ratio=3.663636
[686313719] . SNAIL[0]=29 scale=0
[686313719] . SNAIL[5]=13
[686313719] . SNAIL[5]=13 scale=0
[686313720] . SNAIL[7]=10
[686313720] . SNAIL[7]=10 scale=0
[... 145 lines cut ...]
[686337358] . LX Colibri/Nano installed (platform=le test=1)
[686337359] . Device C is DISABLED.
[686337359] . Device D is DISABLED.
[686337359] . Device E is DISABLED.
[686337359] . Device F is DISABLED.
[686337359] . RestartCommPorts end @h17:25:11 (NO FIX)
[686337365] . ComPort 2 ReadThread : started
[686338937] . First GPS DATE: 2017-3-6  h17:25:11 (UTC 17:25)
[686338945] ... Retuning dstrangeturnpoint to 88 km 
[686338950] ... Retuning dstrangeturnpoint to 76 km 
[686338954] .... RangeTurnepoint=437/500 (TPrange=76) RangeLandable=278/500 (LNrange=150) retunecount=2 retrycount=0 sinceretry=0s 
[686341767] . NANO
[686341767] . NANO Serial Number 2090
[686341768] . NANO Software Vers.: 2.40
[686341768] . NANO Hardware Vers.: 0.00
[686342306] ... GPS baro source back available h17:25:31 (UTC 17:25)
[686342307] . GPS NMEA source changed to port 1  h17:25:31 (UTC 17:25)
[686347773] . AutoQNH: setting QNH to average terrain altitude=221 m
[686347776] QNH set to 1001.05, Altitude 221m
[686362160] ... DeviceSave </storage/emulated/0/LK8000/_Configuration/DEFAULT_DEVICE.dvc>
[686382119] ... ComPort 2 StopRxThread: Cancel Wait Event !
[686382119] ... ComPort 2 StopRxThread: Wait End of thread !
[686402120] ... ComPort 2 StopRxThread: RX Thread forced to terminate!

crash.

brunotl, 2017-03-06:
fixed by commit 11526146c81cdc628e66cb61e7656da7e579fdc7 and 92e66af809166089cf24485f3a6a52dcf9d99566

closed 2017-03-06 by brunotl

pventafridda, 2017-03-06:
ok great thx