LK8000/LK8000 #1073: Topology labels encoding

ToninoTarsi, 2017-03-20:
If we wand an updated topology OpenStreetMap is the obvious source of data for this. OpenStreetMap  uses UTF-8 for string encoding and support all languages. Now LKMAPS_Desktop ( > 1.0.1.5 )  uses same encoding and  shapefile are encodided in UTF-8 
Shapefile encoding is defined in an auxilary .cpf file. Opening this shapefile on GIS software works file :

[image]

This is an example of japan map. Obviously it  does not work with LK8000 

[japan.zip](https://github.com/LK8000/LK8000/files/854967/japan.zip)

Do you think that modifying msDBFReadStringAttribute(..) to reas cpf file and relative encoding is the correct way to proceed ?

pventafridda, 2017-03-20:
can't we just transcode to existing supported format?

ToninoTarsi, 2017-03-20:
At the moment only ASCII is supported ... but I can be wrong

brunotl, 2017-03-20:
current Topology charset is Latin1 (ISO-8859-1),  cf. XShapeLabel::setlabel(...)

ToninoTarsi, 2017-03-20:
Ok Bruno. And this is the default shapefile encoding . I tryed to transcode in LKMAPS_Descktop but for some reason it was not working 
```
Encoding iso = Encoding.GetEncoding("ISO-8859-1");
Encoding utf8 = Encoding.UTF8;
byte[] utfBytes = utf8.GetBytes(Label);
byte[] isoBytes = Encoding.Convert(utf8, iso, utfBytes);
string newLabel= iso.GetString(isoBytes);
```

I will make some other tests.

pventafridda, 2017-03-20:
also accented are in use. I don't remember now how they were transcoded

ToninoTarsi, 2017-03-20:
Going deeper into the problem it looks like :

> "ISO 8859-1 is not a Unicode charset, simply because ISO 8859-1 does not provide encoding for all Unicode characters, only a small subset thereof."

So I think this is why a simple transcoding does not work.  If we what all languages we need a Unicode like UTF-8 support ... maybe :-)

pventafridda, 2017-03-20:
Note that we only use uppercase for topology.
Test in Germany how it looks like now ( no complaints received).

pventafridda, 2017-03-20:
just don't drop iso 8859 support. because people use 8859 maps as a matter 
of fact now. if we can support both, everybody happy.

brunotl, 2017-03-20:
iso-8859-1 covers German, so it's wrong exemple.

japan map available on LK8000.it don't use kanji for topology label, it's why nobody complain.

ToninoTarsi, 2017-03-20:
Looking at the code Bruno quoted it looks like that UTF-8 is already present but it is not activated by the Shapefile encoding ....

```
#ifdef UNICODE
    // from Latin1 (ANSI) To UNICODE
    int nChars = MultiByteToWideChar(CP_ACP, 0, src, -1, NULL, 0);
    if (nChars) {
        label = (TCHAR*) malloc(nChars * sizeof (TCHAR));
        if (label) {
            MultiByteToWideChar(CP_ACP, 0, src, -1, label, nChars);
        }
    }
#else
    // from Latin1 (ISO-8859-1) To Utf8
    tstring Latin1String(src);
    tstring utf8String;

    Poco::Latin1Encoding Latin1Encoding;
    Poco::UTF8Encoding utf8Encoding;

    Poco::TextConverter converter(Latin1Encoding, utf8Encoding);
    converter.convert(Latin1String, utf8String);
    size_t size = utf8String.size();
    if(size) {
        label = (TCHAR*) malloc(size * sizeof (TCHAR) + 1);
        strcpy(label, utf8String.c_str());
    }
#endif
```

brunotl, 2017-03-20:
if you want allow to use utf8 encoded topology you need : 
1.   detect or read from topology file what is encoding.
2.   for ux target (not unicode) : if source is utf-8 do nothing otherwise convert from latin1  to utf-8.
3.   for unicode target : if source is utf-8 convert from utf-8 to widechar otherwise convert from latin1 to widechar

ToninoTarsi, 2017-03-20:
Yes. Exactly. This week I have some days off in the alps but next week I will try.

closed 2017-04-27 by ToninoTarsi