LK8000/LK8000 #1077: Crash on ComPort WriteString

pventafridda, 2017-03-30:
0  libLK8000.so                                       0xa3dee45a ComPort::WriteString(char const*)
1  libLK8000.so                                       0xa3df50c5 EWLinkTimeout(DeviceDescriptor_t*)
2  libLK8000.so                                       0xa3df0d33 devLinkTimeout(DeviceDescriptor_t*)
3  libLK8000.so                                       0xa3df018d ConnectionProcessTimer(int)
4  libLK8000.so                                       0xa3d70c65 ProcessTimer()
5  libLK8000.so                                       0xa3d392b7 WndMain::OnTimer()
6  libLK8000.so                                       0xa3d3c419 TopWindow::RunEventLoop()

BOOL EWLinkTimeout(PDeviceDescriptor_t d){
  (void)d;
  if (!fDeclarationPending)
    d->Com->WriteString(TEXT("NMEA\r\n"));
  return(TRUE);
}

// this is used by all functions to send data out
// it is called internally from thread for each device
void ComPort::WriteString(const TCHAR * Text) {
#if TESTBENCH && (WINDOWSPC>0) && COM_DISCARD
StartupStore(_T("... ComPort write discarded: <%s>\n"),Text);
return;
#else
    int len = _tcslen(Text);
#ifdef  _UNICODE
    int size_needed = WideCharToMultiByte(CP_ACP, 0, Text, len+1, NULL, 0, NULL, NULL);
    char* szTmp = new char[size_needed];
    len = WideCharToMultiByte(CP_ACP, 0, Text, len+1, szTmp, size_needed, NULL, NULL);
    if(len>0) {
        // WideCharToMultiByte return size off Buffer, so trailling '\0' included...
        --len;
    }
#else
    const char* szTmp = Text;
#endif
    // don't write trailing '\0' to device
    if (len > 0) {
        Write(szTmp, len);
    }
#ifdef _UNICODE
    delete[] szTmp;
#endif
#endif
}

pventafridda, 2017-03-30:
Probably on android bluetooth loss of connection with device.

pventafridda, 2017-04-03:
I think it has been solved by recent commits. confirm bruno?

brunotl, 2017-04-03:
maybe, can't be sure without replicate

closed 2017-08-24 by brunotl