LK8000/LK8000 #1096: Overlay configuration

ToninoTarsi, 2017-04-27:
Custom overlay configuration is a very nice and important feature you have introduced in v6.
Anyway the way users have to act for making a custom configuration is not so linear and easy.
Wouldn't it be better to direct configure the overlay in page13/Overlay with an option like 

ON
OFF
AA DELTA Time
AADistance Max
....

eliminating the AUX  Infobox relation ?

pventafridda, 2017-04-27:
Absolutely yes. It is my priority for next versions. After that I will 
start working on new Menu buttons and it will take a lot of time.
Right now we have been working fixing bugs and we have some delay but 
that's normal.

ToninoTarsi, 2017-04-27:
Ok Paolo
Thanks

brunotl, 2018-12-11:
fixed #1215

closed 2018-12-11 by brunotl