LK8000/LK8000 #1097: Direct upload to xcontest

rcoh, 2017-05-04:
It would be awesome if LK8000 could use my device wifi to upload the IGC files directly to xcontest like XCTrack does. I'd be willing to put in the dev work if someone points me in the right direction.

ToninoTarsi, 2017-05-05:
At the moment the main issue I see for this is Curl. We do not jet have this library in out master branch. We have it in some branch but with some issue not yet solved.
**If we want XContest upload Curl is needed.**  For the API point of view ( with Curl ) I do not see any problem. I already have it on my Android APP ( TTLiveTrack24 ) and I can share/port some java code. Obviously an API key will be necessary from XContest/Jakub.
Anyway .. first of all we have to decide if such a feature is important on LK8000.

pventafridda, 2017-05-05:
Same story for OLC. Uploading automatically is a nice feature . But it must 
be accomplished on a general approach Base, not only for xcontest.

AlphaLima, 2017-05-05:
Much more important in my eyes is the download of "Up to Date" waypoint, 
airspace, Flarmnet Files like we already do with the maps.
One of the left big advantage of XCS for the unexperienced users.
Nobody wants to copy files manually anymore these days.

pventafridda, 2017-05-05:
where from? apart from flarm net which is an official source, what else for 
worldwide ?

AlphaLima, 2017-05-05:
http://www.segelflug.de/vereine/welt2000/index.html.en
or even better, but not complete yet:
https://openflightmaps.org/live/

AlphaLima, 2017-05-05:
https://www.openaip.net

pventafridda, 2017-05-05:
those we know already . we need a general approach we cannot make custom 
downloads for each source. Or simply open a Web interface on those sites 
but users than need to figure out where are the downloaded files and to 
move them in place which is much easier with a pc.
And they can already do it if they want, from the smartphone browser.

AlphaLima, 2017-05-05:
Or we can have out own cloud like XCS, StrePla, SeeYou where we copy and 
maintain our own files.
I would take care of: D, NL, AT
Bruno: FR
Paolo: IT
Tahts for almost 60% of the users.
And we will find other user to commit and maintain these Data. GIT would 
be a perfect platform for such data.

rcoh, 2017-05-05:
Definitely agreed about the ability to download airspace / maps automatically. That was a really nice feature on XCTrack that I missed when setting up LK8000. However, you only have to do that once or twice and you need to get your tracklog off the device repeatedly. I've split resource downloading into a separate issue.

pventafridda, 2017-05-05:
That's an additional job to do and it is needed for all countries not only 
for ours. But I agree it is a solution. It is quite easy to set up but 
finding people and coordinate them requires a spare person not us. We can 
discuss it on the forum I am pretty sure we can find somebody.

rcoh, 2017-05-05:
Split to #1098

brunotl, 2017-05-11:
upload to XContest require Secret API key, this can't be put inside our source code...

inside API doc, you can read this :
> Shared secret will be a hidden part of your application and should not be shown, sent or be accesible to the users

[API_gate_flight_documentation.pdf](https://github.com/LK8000/LK8000/files/993055/API_gate_flight_documentation.pdf)

brunotl, 2017-05-11:
good news for french pilot, cfd upload Api don't have this requierement ;-)
http://parapente.ffvl.fr/description_cfdAPI

ToninoTarsi, 2017-05-11:
Yes .. this can be a problem Bruno. Moreover .. I was looking at ContestMng.cpp and I think we have bigger problem than this with XContest. I think we do not have yet and implementation of "FLAT Triangle" distance calculation ...

lshachar, 2017-10-05:
to answer the OP question: you can already achieve this:
you can tell livetrack24 to upload your tracks to different websites, including xcontest, and many others. (you do that on their web interface)
you should also enable live track in lk8000, which is under pilot configuration.

robertgrubba, 2018-04-13:
I'm developing polish leonardo xc server instance (available at https://leonardo.pgxc.pl) and I can provide you documentation that will allow you to build upload mechanism to all leonardo based plaforms (as FlyMe currently does). If you are interested on details please let me know.

robertgrubba, 2018-06-14:
?

brunotl, 2018-06-14:
for sure we are interested, but i'm to busy for work on now ...

ToninoTarsi, 2018-06-14:
Meanwhile you can use my Android APP ( TTLiveTrack24 ) that have support for both LeonardoXC and XContest :-)
Porting to LK8000 will be too much work for me too at the moment and we should also evaluate if this is  such an important feature for a tactical flight computer.

aza1978, 2020-02-29:
@robertgrubba - Would I be able to get the leonardo upload details from you at all?

robertgrubba, 2020-02-29:
https://leonardo.pgxc.pl/doc/direct_upload.html will it be enough?

aza1978, 2020-03-01:
Hi robert, yeh that info is pretty decent... do they have any other api style services for validating login etc. or that's pretty much the extent of it?

cgarciaq, 2022-06-22:
This feature would really be awesome, just saying... Right now, having to upload my IGCs during a trip is very unconvenient using a Kobo (impossible without a PC). I know I can use XCtrack to do so, but discharging two devices during a trip is unnecessary and if doing vol vivac, not possible either.

j2l, 2022-07-16:
I'd also need wifi setup (beyond the ON/OFF button), since GPS and vario NMEA messages are sent over TCP in my case.
It works on Android, but not on Kobo.