LK8000/LK8000 #1098: Download airspace / maps / waypoints from the web to simplify setup

rcoh, 2017-05-05:
Discussed in #1097, this would significantly simplify configuration for new users

ToninoTarsi, 2017-05-11:
I think Maps problem is already solved.
For Waypoint I can easily add it to LKMAPS provided someone makes a list of links to use. Unfortunately I can not help for that list. In HG waypoints are not so stable and vary  very often. If I got that list for Gliding  I will add then immediately,

AlphaLima, 2017-05-11:
I think GIT is already the best platform for that,
here I made some initial data,
would that be OK for you?

AlphaLima, 2017-05-11:
https://github.com/AlphaLima/LK8000_Data

ToninoTarsi, 2017-05-11:
Yes ... something like this can work ..

AlphaLima, 2018-11-08:
Thanks to Toni:
https://play.google.com/store/apps/details?id=org.lk8000.lkmap

closed 2019-01-19 by brunotl