LK8000/LK8000 #1100: BlueFly vario QNH problem

pventafridda, 2017-05-13:
We have report that changing QNH using bluefly vario does not work properly.

Our friend Greg reports:
So after much testing on various android devices both with internal barometer and gps and also with bluefly vario connected I have come to the conclusion that there is quite a major issue with regards QNH and hagl.

The issue is not constant so is somewhat difficult to pinpoint but you may know where to look. 

If the actual QNH reading is below 1013.25,  or close to it ie 1014 and even sometimes 1015,  everything works as it should. The QNH is established and stays correct and the hagl parameter is correct. However,  QNH readings above 1013.25 become more problematic the higher the reading. For instance when lk8000 is turned on and the QNH is 1018 or 1020,  the message shows the QNH reading with the correct hagl and immediately reverts back to 1013.25 which then puts the hagl below ground by as much as 50 meters. Trying to change it manually does not work and it always reverts to 1013.25. This happens whether the devices using internal barometer or connected to bluefly. It happens on every android device I have tested it on. Sony,  Samsung,  LG and Huawei. If navigate by gps is selected, the hagl parameter still shows the Height as if the QNH is 1013.25 when the QNH  reading is actually higher.

On the same devices with the same settings,  on the other flight computer apps this does not happen.

ToninoTarsi, 2017-05-13:
With internal GPS and pressure sensor (LG G3) I never had this problem  ... I just did some other test and today QNH is 1018 as reported to be problematic. Non problems. I will test more carefully but I can not see any issue.

pventafridda, 2017-05-13:
The topic ia about BlueFly vario, that has a very strange driver.
Maybe QNH setting is not done correctly.

pventafridda, 2017-05-13:
The problem with the BlueFly vario is that the device driver is written without using the AltitudeToQNHAltitude function.
Same is inside device driver for CProbe. FlyNet , I guess they have the same problem.
All other device drivers are using the QNH function and work correctly.

brunotl, 2017-05-14:
AltitudeToQNHAltitude parameter are altitute in meter, BlueFly, Flynet, CProbe and some other device report altitude in hPa (static pressure), so AltitudeToQNHAltitude has nothing related with that.

PS : in first message of this topic, i can read "on various android devices both with internal barometer and gps and also with bluefly vario", so it seem's problem if problem exists are not only with bluefly vario....

pventafridda, 2017-05-14:
Pressure set on android internal barometer could have exactly the same 
problem. But I cannot check today. All devices going pressure use 
Altitidetoqnh . I did that long time ago to fix the auto qnh problem if I 
remember correctly.

brunotl, 2017-05-15:
if problem exists, we will see where is in Runtime.log ...

brunotl, 2017-05-15:
bug found : regression commit 834552828b5209f68e4258c179a84e3797bba006

pventafridda, 2017-05-15:
argh that commit was fixing a problem .. that now exist again.
we have to check

closed 2017-06-07 by brunotl