LK8000/LK8000 #1106: WiFi on Kobo Glo HD

ToninoTarsi, 2017-06-01:
After last Kobo Glo HD firmware update LK8000 is not more able to connect to any WiFi hotspots.
As suggested by Bruno issue is similar to : https://github.com/koreader/koreader/issues/2554
I have a friend affected by this and just making some tests adding -Dwext to wpa_supplicant.

brunotl, 2017-06-01:
it's seem's problem is linked to kobo firmware update from version 4.2.8432 (1/9 Feb 2017), so all model can be concerned excepted Mini.

ToninoTarsi, 2017-06-01:
Adding "-D wext" to wpa_supplicant solve the problem on the GloHD with new Firmware and also work on  GloHD with old firmware.
I also tested  Kobo GLO device and works fine.
Left to test are Aura,mini  and Touch devices.

brunotl, 2017-06-01:
Tested on Mini and Touch 2 and Aura 2  that work.

ToninoTarsi, 2017-06-01:
Great!!!

closed 2017-06-07 by brunotl

j2l, 2022-07-16:
Hello,
@ToninoTarsi would you please explain how to add "-D wext" to wpa_supplicant ?
I recently run LK8000 on ClaraHD but I miss wifi for GPS and vario data.