LK8000/LK8000 #1118: Replacing Task waypoing with an optimized waypoint (!waypoint)

parapenT1sta, 2017-06-14:
Using PG mode with Optimized Route ON.

Create a task with 3 waypoints, for example.
Select the first waypoint, then press Select and then replace it with the waypoint with exclamation point ! (optimized waypoint). Then close the dialog.
The name of the waypoint will be replaced with !!!!!!!!!!!!.

Should we block the chance to select the optimized waypoint to avoid issues? Or is this a bug?

pventafridda, 2017-06-14:
looks like a little bug . not sure.
A single goto (1 point task ) must be allowed to use optimised waypoint.
but creating a real task I don't know.
or better I don't remember

brunotl, 2017-06-15:
very old bug, but real bug ...

we don't allow to create Task using "virtual waypoint", so "Task button" in dlgWaypoint need to be hidden.

brunotl, 2017-06-15:
same problem with dlgWaypointDetails : WndFrame "frmCommands" need to be hidden

closed 2019-09-30 by brunotl