LK8000/LK8000 #1119: Move waypoints in Task Edit

parapenT1sta, 2017-06-16:
Situation 1 with these Task Settings in System Setup: 
[image]

Create a task with 3 waypoints, if I move the middle waypoint up to start position or to finish it doesn't move the radius info, only the waypoint. Is this normal? I thought it should move all the info that we setup while introducing the waypoint in the task.

Situation 2 with these Task Setings in the System Setup:
[image]

Create a task with 3 waypoints. 
Start: Cylinder, 2Km
Turnpoint: Cylinder, 1Km
Finish: Cylinder, 0.4km

When I move up the finish it converts the waypoint from Cylinder to Circle ESS.

brunotl, 2017-06-21:
need refator Start and Finish radius storing for AAT and PG optimized task

parapenT1sta, 2017-06-21:
@brunotl Does it make sense having Task settings in the system setup because every task is different every day, at least for us paragliding pilots.

What is your opinion about it?

brunotl, 2017-06-21:
yes it's make sens, because it's unique setting value for default glider task, and it's default value for all other ( AAT or PG Optimized )

pventafridda, 2017-06-21:
same for gliders. but then every task parameter must be exposed in more 
than one page in task setup.

AlphaLima, 2019-07-26:
https://github.com/LK8000/LK8000/issues/1357

parapenT1sta, 2019-09-07:
@AlphaLima The version you sent me doesn't fix the second situation that I mention here. I am using Paragliding Mode. It doesn't happen all the times and It happens more when I have Optimize Route On.

Create a task with 3 waypoints.
Start: Cylinder, 2Km
Turnpoint: Cylinder, 1Km
Finish: Cylinder, 0.4km

When I move up the finish waypoint to the middle position, it converts the waypoint from Cylinder to Circle ESS even when Sector is defined as "Line" in the system setup.

parapenT1sta, 2019-09-07:
@AlphaLima Regarding the first situation it's also the same but maybe it's how LK8000 works. It's only changing waypoint ID but not the whole setup (type of waypoint and size) and maybe make senses.

I will close this topic, but just think about the second situation I reported.

closed 2019-09-07 by parapenT1sta