LK8000/LK8000 #1125: change the way LK8000 calculate tasks to the finish ring

lucaspirolla, 2017-07-24:
it's now pretty standand to have a circle perimiter as finish line, and also a minimum height at this perimeter.  From "Local Procedures"  of WGC 2017 Benalla:
7.7.2 Finish options to be used
The finish will be a Finish Ring of Radius 3 km or as specified at briefing.
7.7.1.a 	Minimum altitude for the finish ring
The minimum height for crossing the finish ring is 300 ft AGL or as specified at briefing.
( http://www.wgc2017.com/useful-info/documents.aspx?contestID=25780 )

So this is fair to assume that LK8000, in order to remain competitive, should calculate all tasks parameters to this finish ring, even if that will mean that the glider will arrive at the airfield with a lower height than the one set in the safety height.  I've been told that other pilots have this option in softwares like SeeYou Mobile.

pventafridda, 2017-07-24:
you can do it without any problem since ever.
Use a waypoint with the specified minimal height.

brunotl, 2017-07-24:
curently, altitude of center is used, not the altitude of crossing point of radius ...

pventafridda, 2017-07-24:
I don't see the difference . if the center is at 300mH you must cross the 
borders over 300mH. No?

brunotl, 2017-07-24:
in mountain with 3km radius that make huge difference ...

lucaspirola, 2017-07-25:
The point is that pilots wants to NAIL 301m while crossing the ring perimeter.  Also, all the distances and tasks statistics are being calulated to the center, not the ring.  Even the Distance Remaining appearing on the top of the screen is showing to the center.
No one is going to the center _at this point_.  Pilots are going to the center only after using all energy to increase TaskSpeed and cross the finish ring at 300m+as little as possible sharp.

And that's can even be used to make final adjustments on the flight.  Suppose you've just advanced your task and now you're on your final glide to the finish ring, and suppose that the path is really good and even  flying at your selected MacCready speed, some factors gave you more energy than expected.  You hit "optimise" (that at this point should have an option to be automatic, i.e., being calculated in "real-time"), and now you see that you better cross the finish ring on another point, not straight to the center point.

The way it is today is not optimal, and we've been struggling with this limitation.

lucaspirola, 2017-07-30:
LX systems will also do this.  From their manual:

"The **Navigate to nearest point** option is very useful and actually a must if a finish cylinder is used with significantly large radius.  Enabling this option will calculate the final glide to the edge of the cylinder rather than to the centre of the cylinder."

As to set it to 300m, they present 2 ways of doing that: one is changing the RESERVE parameter, other would be to raise the altitude of the finish.

pventafridda, 2017-07-30:
solution could be to adopt optimised destination. The waypoint with a 
leading exclamation mark is the optimised one. Example T>Name  and T>!Name

So we don't have to make a revolution in the code and we simply add more 
calculations for optimised waypoint and let people decide on the fly what 
to adopt.
I don't remember how it has been made for paragliding.

pventafridda, 2017-07-30:
I must add that in no case we shall ever consider changing the safety 
parameters for landing places in Lk. People are going to kill themselves if 
we do it . In no case a landable should be used as a task destination . If 
people want to fly without safety parameters they can do it already by 
disabling them. So the arrival altitude at the ring instead of at the 
center has nothing to do with landables and reserve altitude: if Lx does 
that it is a deadly issue for pilots.
This issue has to be considered separately. But I trust that it is enough 
to set up the task as it is now. a waypoint is added in the task with a 
certain altitude. This altitude is then used both for the center and 
fornthe  perimeter in case the optimised destination is chosen.

parapenT1sta, 2017-07-31:
I think @brunotl did most of the work on this commit: https://github.com/brunotl/LK8000/commit/ef26309ae181c02197f3589823350f1b1c38ef77

pventafridda, 2017-07-31:
Yes but we need to keep AAT for gliders.  

From: Sérgio Da Silva 
Sent: Monday, July 31, 2017 12:33 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Comment 
Subject: Re: [LK8000/LK8000] change the way LK8000 calculate tasks to the finish ring (#1125)

I think @brunotl did most of the work on this commit: brunotl/LK8000@ef26309

parapenT1sta, 2017-07-31:
The commit has these options:
+# New Task Type
 +_@M1902_ "AAT"
 +_@M1903_ "Grand Prix"
 +_@M1904_ "Race To Goal"
 +_@M1905_ "Default"
 +_@M1906_ "Task Type"

parapenT1sta, 2017-07-31:
As far I can see Glider mode has 3 types; Default, AAT and GrandPrix.

pventafridda, 2017-07-31:
no. only normal and aat.

parapenT1sta, 2017-07-31:
With that commit I shared? 

[image]

parapenT1sta, 2017-07-31:
Or maybe I am confused. Sorry.

lucaspirola, 2018-03-01:
Will the behavior change in the new version?

@pventafridda when the task has this kind of finish ring (300m @ 3km radius) it's because this is the safety margin already.

It would be great to have this flexibility.

pventafridda, 2018-03-01:
hopefully we will enhance task system for gliders.

lucaspirola, 2018-05-09:
Hello, it's been a while now since I started this thread.
Ventafridda, is the opmization to T> !Finish, optimising !Finish as the ring with 300m possible, and !Finish living in the geometric space of a circle line 3km from waypoint Finish?

mzeddd, 2023-11-15:
Hi,
I just want to lift this issue up.
I'm flying Condor tasks where minimum finish alt is set. 
For example finish alt is set to 800MSL, which means that finish TP is approx at 600MSL and 200m for safety.

If I open LK and (sorry for that) XCSoar I see this difference in calculation. LK is telling me to get 800m more because I set minimum finish alt to 800m. The same settings in XCSoar give me number of meters to climb to fly just over the minimum alt crossing finish cylinder.

Finally, when I'm about to cross finish cylinder 5m above minimum, LK keeps telling me to get extra 800m at the same time XCSoar shows that I have +5 meters (which is correct)

[image]