LK8000/LK8000 #1128: Feature Request: Contour lines on map

pioneer, 2017-07-28:
For E-Ink devices especially, it'd be great to have contour lines on map instead of greyscale colouring, seems it should improve visibility

[image]

pventafridda, 2017-07-29:
Contour aka ISO lines can be made out of raster terrain and they should be 
painted like normal topology (shapes).
Furthermore we should then convert each raster map into shapes . Long work.
I did it for my usage 10 years ago on another navigator. Iso lines were 
making the screen very cluttered. I don't think it is a good idea..

pioneer, 2017-07-30:
Looks not bad on my Kobo, compare with the same on LK8000

[image]
[image]

pventafridda, 2017-07-30:
looks not bad because you have nothing else painted on. Try painting  
topology lines, glide line and task lines too.

cloudsun9, 2018-05-02:
Active until maximum selected zoom level would make this a nice feature.
Would be nice to have it.

brunotl, 2019-10-20:
done

closed 2019-10-20 by brunotl

cloudsun9, 2019-10-23:
Thank you Bruno, this is good.
Might seem fussy but, for e-ink it would be even better as an alternative to the shaded terrain on a white background and (for low lands at least) more lines.
I do appreciate it would be a big effort for a small use case though.