LK8000/LK8000 #1131: DEFAU

roeoender, 2017-08-06:


pventafridda, 2017-08-06:
that's because there haven't been user changing this file in 7 years. Our users are not software engineers.. We have kept all options available in case things require special configurations but so far it was never requested. Initially this file was an internal resource not even available on the file system.I moved it to file system years ago in order to save precious memory in windows ce limited 32M space. It is possible that it comes back internal and no more on the file system. This in fact would solve rollback problems when changing lk versions.

roeoender, 2017-08-06:
Hi, I found this problem when used this file to add System Setup button right in the first menu (I use setup so often that it is more convenient to have it here). I can provide patch / pull request for this file If you want.

It could also be convenient to use this file for experimental config like custom colors of snail trail.

External config file is convenient for "power users" - it would be nice to be able to setup more options with it - like things visible on standard navbars.

Regards,
Jarek Wosik

brunotl, 2017-08-24:
changing menu file is only a workaround for bug in menu managment.

closed 2017-09-04 by brunotl