LK8000/LK8000 #1142: Terrain under sea level is painted all blue

lshachar, 2017-10-05:
I live in Israel and we fly in many areas that are below sea level. unfortunately they are all painted blue, either by using [lk8000's israel online maps](http://lk8000.it/lkmaps/ISR.html) , or by making a custom map with [LKMaps_Desktop](http://www.vololiberomontecucco.it/LKMAPS_Desktop/LKMAPS_Desktop.exe).(!)
This has been discussed by ryszard_lewandowski [here](http://www.postfrontal.com/forum/topic.asp?whichpage=0.533333333333333&TOPIC_ID=8665&#69909) (making his own maps, not for Israel - problem does not only relate to Israel maps) and by me [here](http://www.postfrontal.com/forum/topic.asp?TOPIC_ID=8651) (either my maps or the online maps)

how it appears in lk8000:
[image]

True water areas in Israel:
[image]

The problem is already visible from the [mapset area image](http://lk8000.it/lkmaps/ISR.html) : 
[image] 
(The entire area inland is blue)

I think that the problem is, the LKM file (for Israel) does not contain coast_area.* files! so, since it is unknown to lk8000 where the sea ends and land begins - everything below 0 altitude is painted in blue.
(Netherland's LKM map, for instance, contains three coast_area files: .dbf, .shp, .shx)

Unfortunately LK8000_Desktop Only creates .LKM file rarely for me, (It usually only creates .DEM files) So I can't find a way to create the coast_area files properly, and prove my assumption correct.

__________________________________________________
a bit more info:
Here is a screenshot from QGIS Browser of the water_area.shp inside the topological file LK ISR_ISRAEL.LKM :
[image]
- These are the correct water areas inside israel. no problem there.

I translated the terrain ISR_N_ISRAEL_250.DEM file to an .asc file using [GIS2LKMAP](https://github.com/ToninoTarsi/GIS2LKMAP) which I found out about [here](http://www.postfrontal.com/forum/topic.asp?TOPIC_ID=8475), and it also looks fine:
[image]
- Can't say much but it looks okay to me.

brunotl, 2017-10-05:
> I think that the problem is, the LKM file (for Israel) does not contain coast_area.* files! so, since it is unknown to lk8000 where the sea ends and land begins - everything below 0 altitude is painted in blue.

you think right !

all LKM file without "cost_area.*" need to be rebuild ...

lshachar, 2017-11-02:
Beautiful work, Bruno!

[image]

brunotl, 2017-11-02:
thanks for feedback !

pventafridda, 2017-11-04:
This is not a solution. See OpenCloseTopology lines 106 ..  LKWaterThreshold is my solution.

brunotl, 2017-11-04:
if you have time for check all LKM files, it's your choice, i don't want to do that ( in fact i can't because i don't have tools for generate maps ).

in all case may fix work for general case and also fix few other bug ....

pventafridda, 2017-11-04:
Isachar says:
I think that the problem is, the LKM file (for Israel) does not contain coast_area.* files! so, since it is unknown to lk8000 where the sea ends and land begins - everything below 0 altitude is painted in blue.

Thats not how it works. because otherwise we would have netherlands in blue, for example.  It is exactly the opposite: without coast lines you dont get the blue of the sea, because there is no "sea" in DEM. It is only terrain altitude, and we dont paint blue for terrain unless it is under a parameter called LKWaterThreshold, set by OpenCloseTopology when we load topology. 
Now the problem is simply that without coast areas, we dont set the LKWaterThreshold to -1000.
I have to investigate why Israel has no coast lines, and that's all about this.

pventafridda, 2017-11-04:
okk let me see why Israel has no coasts , first!

brunotl, 2017-11-04:
he say right, that explain why your fix work for Netherland but not for Israel.