LK8000/LK8000 #1154: Running on Raspberry Pi

PhilColbert, 2017-11-30:
Hi,

I have compiled on the PI and running the executable comes back with 

NO LK8000 Directory - FATAL ERROR 000

I have the executable inside the standard directory structure as normal on a PC, tried moving it to /lk as per the nimbus system and /v as in some of the code comments.

No luck yet - any idea?

thanks

brunotl, 2017-11-30:
"LK8000" directory need to be inside "$HOME"

PhilColbert, 2017-11-30:
Thanks, couple of things with the install script.

DEFAULT_MENU.TXT doesnt come over - copied manually
The Dist directory with all the PNG files in doesnt seen to be there anymore, so that fails.

Just working out how to run it full screen now from X :)

brunotl, 2017-11-30:
you build on rPi itself, so you can use  "install" PHONY of make file, that install everything on right place.

PhilColbert, 2017-11-30:
Is there a specific way to run the executable on the Pi ?

Running it from the Windows environment and removing the outer window looks great, its still running windowed, however cpu usage of the executable is bouncing from 20% to 400% in top.

Hence its running quite slowly ?

Any tips would be great, thanks

brunotl, 2018-01-12:
fix : ef2eef01a2ccbd7532e6dd320dc7251eb85ff375

closed 2018-01-12 by brunotl