LK8000/LK8000 #1165: NmeaOut & Ext Sound

PhilColbert, 2018-01-16:
Hi,

Just wondering if Ext Sound works with the NmeaOut device driver?

Thanks

ToninoTarsi, 2018-01-16:
Hi Phil,
I just tested "Ext Sound with LKNet and works fine, Should work I think with any.
Just a question: why do you use a SoundTable in AirWhere and not just stay with the default one ?
Tony

PhilColbert, 2018-01-16:
Thanks, must be bug in my code somewhere, strange :) seems to work by just sending the nmea port valid data outside the kobo.

There wasnt a decent sound table available a long while ago when it was first done , you dont have to use it if you dont want ? :)

closed 2018-01-17 by PhilColbert

brunotl, 2018-01-17:
maybe sync is missed and data are not serialized correctly ...

PhilColbert, 2018-01-17:
Sorry it was my fault , typo in my code :) works great, thanks !