LK8000/LK8000 #1192: Kobo .wpt files issues

villamany, 2018-05-23:
Hi, im using the lk8000 6.0b in my kobo mini, but im having issues with several .wpt files. Whe i use these files is some cases there are no any waypoint loaded.
Lk8000 Android works fine with the same file.

For example this file is fully waypoits missing in kobo:   https://airtribune-production.s3.amazonaws.com/media/contest/waypoints/2017/06/lap2017.CompeGPS.wpt

I also obseved that when working with "good .wpt files" if i use more than 16 chars for the waypoint name the waypoint is also missing when load the file.

Cheers

parapenT1sta, 2018-05-23:
Did you try the latest beta version? Your issue might be fixed already: http://www.postfrontal.com/forum/topic.asp?TOPIC_ID=8737

AlphaLima, 2018-05-23:
have you set "wpt outside terrain" to "include" on Config page 1 ?

villamany, 2018-05-23:
Yes

AlphaLima, 2018-05-28:
PLEASE USE THE LK8000.IT FORUM FOR BUG REPORTS!!!

brunotl, 2018-05-28:
I can't replicate ...

villamany, 2018-05-28:
Are you using the mini for replicate?

brunotl, 2018-05-28:
yes

villamany, 2018-05-28:
I just tried a fresh install again by connecting the kobo to a PC, deleting all files and folders but .kobo directory, copying the "koboRoot.tgz" from the 6.1r to .kobo dir and rebooting, after updating process i copied the linked .wpt file to waypoint directory. But the same result "Waypoints loaded: 0" at startup screen.

brunotl, 2018-05-28:
check RUNTIME.log ....

villamany, 2018-05-28:
here is the log after a fresh installation: https://drive.google.com/open?id=1so6iUgdNuHpDH5ghAvYKn9wizj9LDxvs

villamany, 2018-07-26:
Same issue on the Kobo Aura 2 edition, so i think is a lk8000 related issue.

PD: "Sorry but the lk8000 forum is not working, a lot of issues for registration and logging".

brunotl, 2018-07-26:
in your Runtime.log is see "[000015988] . NO TERRAIN file available."

so check if you have enabled "Waypoint outside terrain" in system config

villamany, 2018-07-26:
Yes, this is because the log was after a clean installation without any configuration. But later tested with that option and the terrain file downloaded and same results. 
If i get the same file but translated to .cup format i see all waypoints as expected without touching any configuration.

brunotl, 2018-07-26:
please send to me your mail,

villamany, 2018-07-26:
villamany@gmail.com

brunotl, 2018-10-01:
@parapenT1sta : 

your file is  "GPSDump/FS" wpt file, this format are not supported.

> files with name of .WPT can be different formats.  both OziExplorer, CompeGPS, and GPSDump/FS uses the same .WPT name on the file, but they are in fact different types of files.

Only OziExplorer and CompeGPS ".wpt" files are supported by LK8000...

parapenT1sta, 2018-10-01:
Sorry! I was using the file given by the organization.