LK8000/LK8000 #1196: Aura second edition hangs.

villamany, 2018-06-22:
Hi, im using the Aura second edition with lk8000 and bluefly ttl board.

If i close the jumper for auto powering on the Bluefly i cant connect the kobo as massive storage, so i don't have any way for download my tracks. 

After several test it seems to be related to koboRx Uart line, if there is data trafic on that line when kobo is booting and usb powered (connected to pc or wall power) the kobo hangs and not boot.

If i disconnect the Rx line then connect to usb, wait some seconds and reconnect the Rx line works as expected.

Is not a power related issue (the BF board still powered all the tme) also tried a direct 5k pull up and a pull down resistor on the Rx line for checking if is a pin level issue when booging but work as expected, only hangs when data trafic is present on Rx line.

Anyone has a the same setup working fine as massive storage?

brunotl, 2018-06-22:
what version of BlueFly module you have ?
before V11 : you need to add 10k pull up resitor on Rx line

villamany, 2018-06-22:
V11 r4 ttl gps (new)

Also got a kobo mini with another v11 and working fine but not on the Aura 2ed

brunotl, 2018-06-22:
we don't have BFV module, we can't test, so you have 2 option :
 - Ask on forum, maybe some one have the answer : http://www.postfrontal.com/forum/default.asp?CAT_ID=11
 - Ask to Alistair : http://www.blueflyvario.com/contact/

villamany, 2018-06-25:
Thanks, but I can't register in that forum. I do not know what I should enter in "Antispam question".

http://www.postfrontal.com/forum/register.asp?mode=Register

villamany, 2018-07-26:
Finally Alistair released a firmware version adding a delayed stream startup that solves the issue, the firmware is for v12 boards but after contact he send me the new firmware and is also compatible with the v11 boards by changing some settings.
I upgrade my v11 firmware and works fine (i use 5 secons delay). 

Now i can still using the autopower on jumper on the bluefly ttl board and kobo Aura 2 edition.

closed 2018-07-26 by villamany

brunotl, 2018-07-26:
thanks for feedback