LK8000/LK8000 #1208: Android Target API Level 26 required after November 1, 2018

brunotl, 2018-10-04:


ToninoTarsi, 2018-10-04:
Yes .. this is an urgent issue to solve. Biside the one that current gradel code does not compile with latest NDK > 17  ( ABIs [armeabi] are not supported for platform. Supported ABIs are [armeabi-v7a, arm64-v8a, x86, x86_64].) .

brunotl, 2018-10-04:
this problem wil be solved after v7 release because too much change are required ...

ToninoTarsi, 2018-11-20:
Everything looks fine for me now. Something left ?

brunotl, 2018-11-20:
it's seem's ndk r18 have "bugge" openmp implementation ( possible conflict with pthread at openmp thread pool creation )

openmp is used for draw terrain and removing them slowdown to much the screen rendering, we need to investigate for right way to solve this problem ...

ToninoTarsi, 2018-11-20:
Ok. I'm using NDK18.1.506 and do not see any problem.. at the moment.

brunotl, 2018-11-21:
most of time that work without any issues, but i have experienced few segfault ....

brunotl, 2019-01-15:
fixed

closed 2019-01-15 by brunotl