LK8000/LK8000 #1213: No "show terrain" option in settings

burakcan, 2018-11-15:
Hello,
I've installed the version 7 on my Kobo and everything works perfect except "show terrain" option is missing from settings. It can be disabled/enabled from the display menu but the setting is lost after restart - as expected.
I believe this is a bug and will be solved sometime but is there a config file which I can manually edit to disable it permanently?

Device: Kobo Aura Edition 2
Version: 7.0

AlphaLima, 2018-11-15:
PLEASE USE THE FORUM FOR SUPPORT!!!
http://www.postfrontal.com/forum/default.asp?CAT_ID=11

If you enable "Save Runtime" on Setup page 22, al lot of settings incl. display settings will be stored persistent.
Background: A lot of users especially newbies wan't  default settings to be resored after restart.

burakcan, 2018-11-15:
Thanks @AlphaLima 
> PLEASE USE THE FORUM FOR SUPPORT!!!
Well, it's actually a bug report + asking for a workaround. I guess github is the place for bug reports?

I don't want the runtime settings to be saved, just want to disable terrain by default. I checked the code but couldn't find the relevant config key.

AlphaLima, 2018-11-15:
So you can,
En-/Disable Terrain in the Display menue
if you want this setting to be stored
enable "Runtime Save"
or
do not select an terrain file

I don't think we need more options, we already have much too much options for the normal user.

burakcan, 2018-11-16:
But the option to disable terrain was there before version 7 I think?

AlphaLima, 2018-11-16:
No!
You probably had "Save runtime" enabled before.
Once again:
Use the forum for such issues!!!

burakcan, 2018-11-16:
Why do you have to use so many "!"s and capital letters?
I'm filing something I think a bug, not a "how do I do this" question.

On the "official website" there's a FAQ section about paragliders/hang gliders: http://www.lk8000.it/faq/31-paraglidersdelta/48-para-best-of.html

In the section 4, it says: 
```
4. Terrain
Terrain display: Off
```

Also in the translation files, you can find 2 different "Terrain display" texts.

So it's either a misinformation on the website or the option was there but now is missing from the software.

burakcan, 2018-11-16:
Ok I've found it. It was removed in this commit:
https://github.com/LK8000/LK8000/commit/becaa50f329ec12320b3dc9de141b9f9a3bb0cac

closed 2018-11-16 by burakcan

burakcan, 2018-11-16:
btw, I've seen complaints from the developers a couple of times that there is not enough interest in the community to help development or documentation.
Now I started to believe that it has to do with the attitude of devs. With this attitude of yours, I would be scared to contribute - or even use the product.

brunotl, 2018-11-16:
Just for remember. All contributor are only volunteers and work on there free time...

pventafridda, 2018-11-16:
I normally don't want to break conversations, but let me say that it will 
be a great pleasure if you leave and dedicate your efforts on some other 
software. Not welcomed here in any case . Your contributions would be 
rejected forever. Never merged. And if you use Lk or not, I can't care 
less. Really.

burakcan, 2018-11-16:
@brunotl thanks for the reminder, I am familiar with how open source and/or libre software works. But I don't understand what it has to do with this?

@pventafridda wow, just wow.

AlphaLima, 2018-11-17:
Just a few words of explanation:
-GIT is the place where the developers discuss coding issues
-it is NOT the place for users to decide what is a bug or not
-this, feature request and discussions can be made in the Forum, which is the place you most probable can already find an answer to questions, features request and known bugs.
-the decision if it is a bug or a missing feature, the order of fixing bugs and adding features is made by the developers.
- especially adding more configuration options is critical, for an average user it is already too much configuration possibilities. So this is not a bug it was removed by intension in 2012 by intension!
-the chances of getting a bug fixed or a feature added drastically drops in case of impolite or rude requests, or ignoring the rules.
-Even more if still ignoring after getting a hint like in this case.
We develop this software in 1000s of hours in our free time (often stealing from our families) to offer you a complete free software. The minimum we can ask for is to follow this rules.