LK8000/LK8000 #1238: Multiple vario sources conflict

elgarrapata, 2019-01-10:
When having more than one device feeding vario I can clearly see LK displaying and processing both sources, instead of considering only the 1st device configured.
My setup:
.Android LK8000
.1st device - IOIO uart GPS module
.2nd device - Bluetooth XCTracer mini II GPS (LK8EX1)
.3rd device - IOIO uart GoFly Picospeed (LK8EX1+PLKAS)

Cheers!

AlphaLima, 2019-04-02:
Of course,
priotizing is implemented for GPS and Baro only. 
Two varios connected does not make sense.

elgarrapata, 2019-04-02:
Yes, but the catch here is that I have a device outputting both LK8EX1 and
also PLKAS, and other one with only LK8EX1, which happens to be a better
vario, and I still need that airspeed..

brunotl, 2019-04-02:
workarround  : if you use Generic driver for DeviceC ( GoFly PicoSpeed), you will have aispeed from PLKAS and vario from LK8EX1 will be ignored

AlphaLima, 2019-04-02:
Normaly the sentences should be configurable in the vario device.
However for the future I'm working on configurable drivers:
https://github.com/LK8000/LK8000/pull/1308
But thats a long way to go!

elgarrapata, 2019-04-02:
Ok, many thanks. The GoFly alternative config doesn't feature the airspeed.

closed 2020-06-05 by brunotl