LK8000/LK8000 #1302: LK8000 with Bluefly TTL_v11

alexandre-pereira, 2019-03-25:
Hello,

Just tried to update my version 6.00B of LK8000 to 7.0 (latest) and I actually have a lot of issues with the GPS in the app displaying "GPS NO VALID FIX" sometimes (not all the time) during a flight, cutting my IGC tracklogs in multiple files (I also checked those files and they are sometimes under ground level).

I tried multiple configs (LX something in Device A setup, Blueflyvario as well, same with the baudrates etc), nothing worked.

I went back to 6.00B and everything worked as expected again.

Config : KOBO MINI + BLUEFLYVARIO/GPS V11TTL (linked directly, no bluetooth nor wifi)

Maybe is that the V11 TTL version of the Bluefly vario is not supported anymore by LK8000?

Thanks for the infos,
Alex

brunotl, 2019-03-25:
V11 TTL version of the Bluefly vario is supported.

can you do a NMEA Log with v7 and send it to me with Runtime.Log and IGC files?

thanks.

alexandre-pereira, 2019-03-26:
[NMEA_2080-01-06-00-00-29.txt](https://github.com/LK8000/LK8000/files/3008159/NMEA_2080-01-06-00-00-29.txt)

alexandre-pereira, 2019-03-26:
[2019-03-22-XLK-PEN 1 TO 24.zip](https://github.com/LK8000/LK8000/files/3008160/2019-03-22-XLK-PEN.1.TO.24.zip)
(During a 100km flight, didn't had NMEA log ON on this one but same bug, don't know if it helps)

brunotl, 2019-03-26:
can you sent to me your RUNTIME.log
your "2029-03-22.." log are very surprising because it's seem's each part end with landing detection ...

PS : we can continue by mail in french if you want...

alexandre-pereira, 2019-03-26:
[RUNTIME.log](https://github.com/LK8000/LK8000/files/3009658/RUNTIME.log)

Here you go :) As you want, I'll send you my email, you can call me as well if you need more infos. Maybe it's faster and better for communication.

brunotl, 2019-03-26:
evrything lokk good in your runtime log but it only contains v6.0b run, so no surprise...

villamany, 2019-03-31:
What firmware do you have on the Bluefly borad?
And what speed are you using on the kobo?

alexandre-pereira, 2019-06-19:
Found the problem.
"Aircraft Setup" was on "Glider" and not "Paraglider/Hangglider", and I was using the barometric altitude, it was a high pressure day, so my track was under the ground due to a bad calibration. That's why I got so many tracks.

closed 2019-06-19 by alexandre-pereira