LK8000/LK8000 #1313: Oracle not seeing cities

airmalik, 2019-04-05:
I'm using v7 on a Kobo Glo HD and getting familiar with it on Windows.

Having some trouble with the Oracle feature. It works as expected in some places but in others it either can't 'see' towns that are visible on the map or ignores large towns that are closer. In some cases it simply apologies "Very Sorry your position is unknown". Is this an issue with the maps I'm using? I've tried AUS_SEQ_QLD.LKM, AUS_NSW.LKM and AUS_KEEPIT.LKM. Here's an example where Kilkivan and Woolooga are visible on the map but Oracle says your position is unknown.

[image]

brunotl, 2019-04-05:
first, cities search is made in a 30km radius circle, not in screen bounds

checking Topology file you have tested, in all of them  all city are "big city", so they have same size.

for replicate we need for each case 
  - Topology file Name
  - Aircraft Position ( Latitude, Longitude )
  - result you get
  - result you expect
  - optionaly a map screenshot in PAN mode

thanks
Bruno

airmalik, 2019-04-08:
Hi,

Below are three examples where the Oracle feature doesn't work as expected. In the first two cases, it can't find the position. In the third one, it announces position from a smaller city farther away instead of the bigger city which is also closer. In all cases, the cities that I'm expecting a distance to are displayed on the map.

Map File: AUS_SEQ_QLD.LKM
Terrain file: AUS_SEQ_QLD_500.DEM
Aircraft position: S26.1907 E152.4203
Expecting: You're 10km South of Woolooga
Actual: Very Sorry Your position is unknown

Map File: AUS_KEEPIT.LKM
Terrain file: AUS_KEEPIT_250.DEM
Aircraft position: S30.4517 E150.6873
Expecting: You're 7.5km South East of Barraba
Actual: Very Sorry Your position is unknown

Map File: AUS_KEEPIT.LKM
Terrain file: AUS_KEEPIT_250.DEM
Aircraft position: S30.3347 E149.7132
Expecting: 7 km West of Narrabri
Actual: 14 km East of the city  <Solar Observatory>
19 km East of <Yarrie Lake>

brunotl, 2019-04-08:
perfect, thanks

closed 2019-04-08 by brunotl