LK8000/LK8000 #1350: Unable to create task with large number of waypoints.

RickvdSloot, 2019-05-17:
On a PNA (Vertica V2) you are limited to the number of waypoints you can add.

In landscape mode the maximum number is 8 waypoints, in portrait mode the number is somewhat higher.

During adding waypoints there is some automatic scrolling active in the menu.
It seems after selecting new waypoint the list scrolls up to the beginning and then activate the 'button'.
With 8 waypoints in the list you first have to scroll down to be able to select 'add waypoint' but after selecting the function you end up with the list at the top and the 6th waypoint selected.
Scrolling back down gets you back into the same loop.

AlphaLima, 2019-05-17:
Please report such issues at the LK8000 Forum, so every user can help and profit from it!
(GIT is for developers communication)

which version?
Have you tried the latest beta? As far as I recall this has been fixed already.

AlphaLima, 2019-05-18:
Confirm! (PNA Version only)

Please use the LK8000 forum for bug reports!

closed 2020-02-25 by brunotl