LK8000/LK8000 #1357: Task definition TP diametr problem after inserting new TP or saving the task

polachp, 2019-06-18:
Hello,
me and also few pilots have come to problem while defining competition task.

If you change the TP radius, its messes up the radiuses of other turnpoints in the task. (it looks like to me that it offsets the radiuses array so that TP2 radius become TP1 radius, etc)
This doesnt happen every time, I dont have precise replication, but I will try to find it.
Mostly appears when you also save the task, and if you return back to the task definition screen, TPS radiuses are wrong.

AlphaLima, 2019-07-26:
RACING:
For racing tasks, there are only three radius/type properties
SART, TURNPOINT, FINISH
All TURNPOINT do have the same radius. If you change the the waypoint order the first will get the START type/radius, the last the finish radius/type and all others the TURNPOINT  radius/type

AAT:
For AAT you can change the the radius for each turningpoint sector individually.
If you change the the waypoint order, the radius/type will move with the turnpoint.

That is all intended!

brunotl, 2019-07-26:
other info :

if you use PG/HG mode and OptimizedTask is on (inside timegates config), radius definition work same as AAT described by Alphalima.

brunotl, 2023-02-21:
duplicate #1554  ?