LK8000/LK8000 #1415: Support OpenAIP airports, airspace... format

scls19fr, 2020-03-01:
Hello,

[openAIP](http://www.openaip.net/) is a web-based crowd-sourced aeronautical information platform which deliver free, current and precise navigational data to anyone.
Supporting their file formats could be a nice improvement.

See http://www.openaip.net/downloads for documentation
http://www.openaip.net/system/files/openAIP_aip_format_1_1_airport.pdf
http://www.openaip.net/system/files/Airspace.pdf
http://www.openaip.net/system/files/openAIP_aip_format_1_1_navaid.pdf
http://www.openaip.net/system/files/openAIP_aip_format_1_1_hotspot.pdf

Kind regards

PS : similar issue https://github.com/XCSoar/XCSoar/issues/355

brunotl, 2020-03-01:
already supported, was added 4 years ago ...

just a tips : careful about OpenAIP airspaces data, it's not official data, it contains lot of error ... ...

closed 2020-03-01 by brunotl

scls19fr, 2020-03-01:
Sorry about that.
Are airports aip data also supported?

AlphaLima, 2020-03-02:
You obviously don't test it yourself!
Yes! However for Waypoints OpenAIP also offers CUP Files.

scls19fr, 2020-03-02:
aip data for airports provide much more information than in cup file.

AlphaLima, 2020-03-02:
We already support OpenAIP for airspaces and Waypoints!!!!
You obviously have not tested it yourself nor work with it.
What are you missing and what is your intension???

scls19fr, 2020-03-02:
Thanks for your welcoming help. It works as advertised... maybe adding runway width should be considered.
I think airports data should be considered differently than simple waypoint and a priority between waypoint data and airport data should be set ie if a waypoint and an airport are closer than 200m only airport should be displayed.

brunotl, 2020-03-02:
- runaway dir and length are already handled 
- turnpoint declutering order is another subject, so open another issues for that.

scls19fr, 2020-03-02:
dir and length are perfectly... Not width which is an available data.

I will open an other issue about turnpoint declutering order

scls19fr, 2020-03-02:
Supporting at least 3 waypoints files will be also a great feature to have.
One for airports one for national turnpoints (France.cup for example) and a last one for personal turnpoints.
Such waypoints organisation will simplify updating data without hassle.

brunotl, 2020-03-02:
please test the last beta...

scls19fr, 2020-03-02:
Thanks for the tip...
9 waypoints is enough!