LK8000/LK8000 #1416: GPS: NO VALID FIX with LK8000-beta 7.1k (Feb 25 2020)

scls19fr, 2020-03-02:
Hello,

I'm trying LK8000 beta but I see "GPS: NO VALID FIX" on map
and position is not updated.

I compared my settings between LK8000 stable and beta and couldn't find what is wrong.

FLY / MENU / Config / Config 2/3 / LK8000 Setup / Device Setup

Device configuration A
Name: Internal
Status: nothing with beta but GPSFix with stable version (7.0 Oct 26 2018)
Geoid Altitude: ON
Serial mode: Normal
Device model: Generic

When I click on Terminal button I see some text which is updating with beta version... strangely when I click on Terminal button with stable version I see "NO DATA RECEIVED" while position is updating correctly at startup!

Hardware: smartphone Huawei P30 Lite
OS: Android 9 with EMUI 9.1.0

Any idea?

Kind regards

scls19fr, 2020-03-02:
I also tried to rename `_Configuration_beta` to `_Configuration_beta_old` to force creation of new config file but same problem "GPS: NO VALID FIX" occured

brunotl, 2020-03-02:
you need to check GGA and RMC sentence  : http://aprs.gids.nl/nmea/#gga

to make NMEA log, use  "menu" -> "Config 2/3" -> "NMEA Logger"
log is writed in folder "LK8000/_Logger",  file name pattern is : "NMEA_YYYY-MM-DD-hh-mm-ss.txt"

brunotl, 2020-03-02:
another tips to avoid "NO VALID FIX" : go outside, GPS is not made for indoor location...

scls19fr, 2020-03-02:
Why it could work with stable version and not with beta?

brunotl, 2020-03-02:
- stable version use android "LocationListener" (depending of device but can be a fused location).
- last beta use only GPS.

scls19fr, 2020-03-02:
ok I will go outside (when weather) will be better and will reopen this issue if it's still failing.
Sorry.

closed 2020-03-02 by scls19fr

AlphaLima, 2020-03-02:
[image]

AlphaLima, 2020-03-02:
[image]