LK8000/LK8000 #1418: Missing Language File with make TARGET=PC FULLSCREEN=y

parapenT1sta, 2020-03-20:
PC version with Full Screen doesn't work with the new language file system

[001607515] ------------------------------------------------------------
[001607515] . [001607515] Starting LK8000 v7.1k Mar 20 2020 PC
[001607515] . Free ram=2097151 K  storage=0 K
[001607531] . Host and Cpu informations not available
[001607546] . Program execution path :   <D:\LK8000\>
[001607546] . Program system directory : <>
[001607562] . Program data directory :   <>
[001607562] ... Missing Language FILE <_Language\en.json>
[001607593] . InitLKScreen: AUTORES 1520x825
[001607671] ... SCREEN RESOLUTION CHANGE DETECTED: 1536 x 864
[001607671] . InitLKScreen: AUTORES 1536x864
[001607796] ... Missing Language FILE <_Language\en.json>
[001611906] . Quick shutdown requested before terminating startup
[001611921] . WndMain::OnDestroy

parapenT1sta, 2020-03-20:
Same error with no Full screen mode.

parapenT1sta, 2020-03-20:
Forget it....!!! My problem! Didn't copy all files! SORRY!

closed 2020-03-20 by parapenT1sta

brunotl, 2020-03-20:
look into makeall script to see how to build distrib archive including all files ...